meshtastic/firmware #3449: last_heard can be 0

jp-bennett, 2024-03-20:
### Category

Other

### Hardware

Not Applicable

### Firmware Version

yes

### Description

when adding a new device to nodeDB, the time on the incoming packet is the only time source used. If the local node has a trusted time source, we should probably use the local source instead.

jimwhitelaw, 2024-03-25:
I had a peek into this and it looks like the code is trying to do as you suggest, but there's still cases where last_heard can be saved as 0. It appears to hinge on the definition of a "trusted time source".  If the time source of the node saving to the nodeDNB is not of quality > RTCQualityFromDevice, it will leave last_heard at 0. Is that the scenario you're encountering or something else?

https://github.com/meshtastic/firmware/blob/728b58fb94f813a11c6aaa3d889749e0df3aa47e/src/mesh/Router.cpp#L441-L445
https://github.com/meshtastic/firmware/blob/728b58fb94f813a11c6aaa3d889749e0df3aa47e/src/mesh/NodeDB.cpp#L894-L895

RTC Quality values:
```C
enum RTCQuality {

    /// We haven't had our RTC set yet
    RTCQualityNone = 0,

    /// We got time from an onboard peripheral after boot.
    RTCQualityDevice = 1,

    /// Some other node gave us a time we can use
    RTCQualityFromNet = 2,

    /// Our time is based on NTP
    RTCQualityNTP = 3,

    /// Our time is based on our own GPS
    RTCQualityGPS = 4
};```

jp-bennett, 2024-03-26:
As @jimwhitelaw demonstrated, the packet time gets overwritten by localtime, making this a non-issue.

closed 2024-03-26 by jp-bennett