meshtastic/firmware #3460: [Feature Request]: Support hmc5883l & qmc5883l

unbelauscht, 2024-03-21:
### Platform

ESP32

### Description

Support hmc5883l & qmc5883l modules would provide accurate compass data for heading vs buggy GPS calculated heading.

garethhcoleman, 2024-04-11:
I can see that QMC5883L has some mentions in the code as a telemetry sensor but it isn't listed as supported (here)[https://meshtastic.org/docs/configuration/module/telemetry/#currently-supported-sensor-types]. Recently @caveman99 made this comment about telemetry for an accelerometer that I think is relevant to your request "We only support slow-changing data or data with is averaged over time. Use cases for this type of sensor is heading while stationary (add-on to position module)".

fifieldt, 2024-09-16:
@dhskinner is working on accelerometer support which should be helpful for compasses 
https://github.com/meshtastic/firmware/issues/4662

for now, a QMC5883L will be detected if plugged in by I2C, but nothing much will happen with it.