meshtastic/firmware #3468: [Bug]: fixed position cannot be set from cli nor android app on 2.3.1.4fa7f5a

saahbs, 2024-03-22:
### Category

Other

### Hardware

Rak4631

### Firmware Version

2.3.1.4fa7f5a

### Description

I used fixed position with previous fw version. after upgrading to this one, I cannot set it via cli nor via android app.

For cli see log output below, for app I set it in gui, reconnect and shows as lat/long/alt of 0.0.

### Relevant log output

```Shell
``
$> meshtastic --setlat 37 --setlon -121 --setalt 30
Connected to radio
Fixing altitude at 30 meters
Fixing latitude at 37.0 degrees
Fixing longitude at -121.0 degrees
Setting device position

$> meshtastic --export-config
...
  position:
    broadcastSmartMinimumDistance: 100
    broadcastSmartMinimumIntervalSecs: 30
    fixedPosition: true
    gpsMode: NOT_PRESENT
    gpsUpdateInterval: 120
    positionBroadcastSecs: 900
    positionFlags: 811
...
```

saahbs, 2024-03-22:
note: the location section is not present at all!

thebentern, 2024-03-22:
Location is not a part of config. It's stored on the node in the NodeDB of the device

saahbs, 2024-03-23:
As of 2.2.24 to which I downgraded, the fixed position is stored in location: which is NOT nodeDB AFAICT, see:

```
$> meshtastic --export-config
...
location:
  alt: 30
  lat: 37.310116
  lon: -121.88448199999999
...
```

This does NOT work on 2.3.1.

francisuk1989, 2024-03-23:
I had this issue aswell on 2.3.1 alpha however did manage to fix it by turning off the fix gps in the android app and then used the python ctl

`meshtastic --host 192.168.2.89 --set position.fixed_position true --setlat XXX --setlon XXXX --setalt XX`

saahbs, 2024-03-23:
Interesting, perhaps with 2.3.x one needs to specify fixed_position true and ALL of lat/lon/alt at ONCE for it to take effect. I did ``--export-config`` from 2.2.24, which had ``fixed_position: true`` and all ``location:`` fields set, upgraded to 2.3.1 again and ran ``--configure exported.yaml`` and it took.

Somehow order of operations matters now and this breaks the android app.

ps. I checked my earlier export-config which I used before 2.3.1 upgrade and it DID not have ``location:`` in it (maybe I was experimenting and cleared it - which explains why I ran into this initially and not after I did full dump on 2.2.24 before configure on 2.3.1.

saahbs, 2024-03-23:
moar debugging on 2.3.1:

```
> /meshtastic --setlat 0 --setlon 0 --setalt 0 --set position.fixed_position false
Connected to radio
Fixing altitude at 0 meters
Fixing latitude at 0.0 degrees
Fixing longitude at 0.0 degrees
Setting device position
Set position.fixed_position to false
Writing modified preferences to device
```

but when I --export-config I see:
```
...
  position:
    broadcastSmartMinimumDistance: 100
    broadcastSmartMinimumIntervalSecs: 30
    gpsMode: NOT_PRESENT
    gpsUpdateInterval: 120
    positionBroadcastSecs: 900
    positionFlags: 811
...
location:
  alt: 30
  lat: 37.310116
  lon: -121.88448199999999
```

so clearly the device is not accepting the lat/lon/alt for some reason. THIS is the bug.

saahbs, 2024-03-23:
This does work:

```
> /meshtastic --setlat 1 --setlon 2 --setalt 3 --set position.fixed_position true
Connected to radio
Fixing altitude at 3 meters
Fixing latitude at 1.0 degrees
Fixing longitude at 2.0 degrees
Setting device position
Set position.fixed_position to true
Writing modified preferences to device
```

and --export-config:
```
...
  position:
    broadcastSmartMinimumDistance: 100
    broadcastSmartMinimumIntervalSecs: 30
    fixedPosition: true
    gpsMode: NOT_PRESENT
    gpsUpdateInterval: 120
    positionBroadcastSecs: 900
    positionFlags: 811
...
location:
  alt: 3
  lat: 1.0
  lon: 2.0
...
```

saahbs, 2024-03-23:
Lets try setting lat/lon/alt without passing set:

```
> meshtastic --setlat 4 --setlon 5 --setalt 6
Connected to radio
Fixing altitude at 6 meters
Fixing latitude at 4.0 degrees
Fixing longitude at 5.0 degrees
Setting device position
```

and --export-config:

```
...
  position:
    broadcastSmartMinimumDistance: 100
    broadcastSmartMinimumIntervalSecs: 30
    fixedPosition: true
    gpsMode: NOT_PRESENT
    gpsUpdateInterval: 120
    positionBroadcastSecs: 900
    positionFlags: 811
...
location:
  alt: 3
  lat: 1.0
  lon: 2.0
...
```

FAIL, passing any of --set{lan,lon,alt} now appears to REQUIRE --set position.fixed_position true even if it is ALREADY set to True.

^^^ please fix.

GUVWAF, 2024-03-24:
You now have to disable fixed position first in order to be able to set your position. This was deliberately put in by #3403, because there have been many reports of people that expected the position to never change after you set fixed position to true. This was especially for phone apps, but the CLI uses the same API.

thebentern, 2024-03-24:
We need to update python to send the position packet first and then enable fixed position as a part of this process. There is no reason to use --setlat --setlon etc outside of the context of a fixed position