meshtastic/firmware #3472: [Feature Request]: ability to only send messages over rf when mqtt is enabled to save on duty cycle

alexderpyfox, 2024-03-22:
### Platform

ESP32

### Description

Currently it's sending out a lot over rf when mqtt is enabled to the point it runs out of duty cycle when you send only one message this is a big issue adding the option to only send text over rf would fix the issue of the duty cycle running out so fast when connected to mqtt

francisuk1989, 2024-03-25:
Or have a option to say when the duty cycle has run out send over MQTT.

fifieldt, 2024-09-16:
There have been significant enhancements to the community MQTT server and the firmware to limit less useful packets. The new default-off OkToMqtt setting should further assist. If things are still bad after your local mesh is on relatively recent firmware please feel free to re-open!

closed 2024-09-16 by fifieldt