meshtastic/firmware #3475: lock/encrypt option for read/write access to physical device

schich02, 2024-03-23:
### Platform

NRF52, ESP32

### Description

It should be possible to passwort lock/encrypt a meshtastic device in such a way that it can not be accessed without a passwort if the physical device gets lost.
Currently if a device gets into the "wrong hands" you must assume everything stored on it is compromised (channel keys, etc.), and unfortunately there is no way of preventing that.

eureekasigns, 2024-03-25:
Interesting idea, but many devices seem expected to communicate directly via a serial connection. 

One workaround that someone posted on Reddit may be of interest. 

Only add an admin channel for a specific remote node to your current client device when you wish to configure it, then remove the channel from your current device that is being used as a client. 
Store the keys in a password safe for each node. 

Example
Node 1 is a router, repeater, or otherwise "remote" device. Add an admin channel. Store the key for that channel somewhere, such as a password safe. 

Node 2 is your "current" device (used for home or mobile use). Only add an "admin" channel for "Node 1" when you wish to manage that device. Otherwise, do not leave the admin channel active for longer than necessary. Only add "admin" channel for other nodes (eg: Node 3, or Node 4) as needed.

There was also mention of only having an admin channel on a router, as it doesn't necessarily need to store messages from any channels, but will still route traffic if "All" is selected. That is another matter and is up to the individual. 

It may be worth adding such notes to some documentation somewhere, as it seems to be a good practice for remote nodes if one wishes to remotely use an admin channel for them. 

Credits: 
Thread https://www.reddit.com/r/meshtastic/comments/1bg0zsh/this_is_how_we_roll_in_the_open_source_country/?rdt=60841
Google doc linked from there https://docs.google.com/document/d/13yf6rwRG-Cv5clap3Lczx88fdKbU5ZJxV89bBW7-MnM/edit?usp=sharing

SpudGunMan, 2024-04-26:
You can set the device in remote management mode and shut off serial and BT preventing any local tampering or detection

closed 2024-10-07 by thebentern