meshtastic/firmware #3491: [Feature Request]: Ambient Light Sensor Module support on RAK

kri164, 2024-03-25:
### Platform

NRF52

### Description

Hello,

just attached RAK12010 WisBlock Ambient Light Sensor Module to my RAK baseboard to get know the ALS module is not supported.

Can you consider to add support for this module in future, please?

This should be the same hardware as Adafruit VEML7700 Ambient Light Sensor.
It is simple sensor, it returns number of Lux, no compute is needed.

[RAK12010](https://docs.rakwireless.com/Product-Categories/WisBlock/RAK12010/Overview/)
[VEML7700](https://learn.adafruit.com/adafruit-veml7700/arduino)

Thank you very much

thebentern, 2024-03-25:
We want to add lux sensors to environment telemetry. Not sure about ambient lighting module, but we can take a look.

kri164, 2024-03-25:
Hello Ben,

same thing, different names I hope.

Thank you

caveman99, 2024-03-25:
This should make it more clear. Not the meshtastic module, just add ambient light level to telemetry and support that sensor ;-)

kri164, 2024-05-23:
Thank you very much.
I will flash next new firmware and will report.