meshtastic/firmware #3492: [Bug]: 2.3.2 instantly maximum transmission duty cycle

francisuk1989, 2024-03-25:
### Category

Other

### Hardware

Heltec Wireless Tracker

### Firmware Version

2.3.2

### Description

Upgraded to 2.3.2 last night and i was sending a message in the default channel and i suddenly got maximum transmission normally i never got this message in the previous builds as im using a standard antenna out the box and only seem to pickup just one node.

So next morning i have tried to send another message again but same message as above and can see my node is ChanUlt 0.0 and ChanTX 0 0

GUVWAF, 2024-03-25:
Do you have MQTT downlink enabled? 

Can you show the serial logs with the [CLI](https://meshtastic.org/docs/software/python/cli/usage/) command `meshtastic --noproto` or any serial monitor on baud 115200?

francisuk1989, 2024-03-25:
> Do you have MQTT downlink enabled?
> 
Yes for MQTT but only set to Uplink only in Channels

> Can you show the serial logs with the [CLI](https://meshtastic.org/docs/software/python/cli/usage/) command `meshtastic --noproto` or any serial monitor on baud 115200?

Give me around 2 hours and i will give send the serial logs but just wonted to say about MQTT.

francisuk1989, 2024-03-25:
> Can you show the serial logs with the [CLI](https://meshtastic.org/docs/software/python/cli/usage/) command `meshtastic --noproto` or any serial monitor on baud 115200?

i have attached a serial logs to the 1st post.

GUVWAF, 2024-03-25:
I don’t see anything special in that log. It shows you tried to send a message, but after a while no one acknowledged it. Is the issue still happening? 
Which which app/client are you connecting to the node?

francisuk1989, 2024-03-25:
> Is the issue still happening?
Not got the maximum transmission  since factory reset yet.

garthvh, 2024-03-25:
Max retransmit means nobody got your message, has nothing to do with the duty cycle.

closed 2024-03-25 by thebentern

francisuk1989, 2024-03-25:
> Max retransmit means nobody got your message, has nothing to do with the duty cycle.

the whole issue was just a sudden message of "maximum transmission" but since i not get that message again then i cant debug it.

never mind.