meshtastic/firmware #3495: [Board]: Affordable, portable 1W Meshtastic ready platform with ESP32+OLED+GPS pads (Radiomaster Bandit Nano ELRS Module)

uberhalit, 2024-03-26:
### SOC

ESP32

### Lora IC

SX1276

### Product Link

https://www.radiomasterrc.com/products/bandit-nano-expresslrs-rf-module

### Description

Being a drone racer, I've always wondered if we could turn some ELRS 900MHz gear into a low-cost, reliable, and super powerful Meshtastic device.

By adhering to the [ELRS board definitions](https://github.com/ExpressLRS/targets/blob/master/TX/Radiomaster%20Bandit%20Micro.json), I was able to incorporate base support for the model referred to above:
[image]
[Fork available here.](https://github.com/uberhalit/firmware/commit/8397551f24b6ee83d6f61de684dcbd1d6046508c)
The display requires a flipping in settings.

Nevertheless, currently, neither of the two PAs appears to be engaged, and the power output is below -20dBm. I attempted to hack together a basic 30dbm output PoC, but was unable to do so. 

[ELRS is calling `dacWrite()` on pin 26 (power_apc2)](https://github.com/ExpressLRS/ExpressLRS/blob/66998e173252c4c0a9e6e0e89010761f56ec9ff2/src/lib/POWERMGNT/POWERMGNT.cpp#L298) and a regular [`writeRegister` to register 0x09 to activate the PAs.](https://github.com/ExpressLRS/ExpressLRS/blob/66998e173252c4c0a9e6e0e89010761f56ec9ff2/src/lib/SX127xDriver/SX127x.cpp#L263)
The [call to reg 0x09 is already being executed](https://github.com/jgromes/RadioLib/blob/9c6eb20169b1019ca4215f6756a7fd32be2649b0/src/modules/SX127x/SX1278.cpp#L336) in the current firmware implementation, albeit with different settings.
Manually setting both to the same values as seen in ELRS source does not seem to work, I'd assume that either the SX1276 support is incomplete or a crucial PA initialization is missing (mostly related to [RFAMP](https://github.com/ExpressLRS/ExpressLRS/blob/3.3.2/src/lib/RFAMP/RFAMP_hal.cpp) or [LR1121](https://github.com/ExpressLRS/ExpressLRS/blob/66998e173252c4c0a9e6e0e89010761f56ec9ff2/src/lib/LR1121Driver/LR1121.cpp#L296)). 

In any case, Radiomaster is known for their excellent support. They immediately responded and offered technical help, so this board should come to live eventually.

dhandzic, 2024-03-28:
Hmm, the SX1276 supports +20dBm and the LR1121 +22 dBm output. Unless there is an external PA it is hard to see how +30dBm could be achieved. Any information if there is an external PA and how it is controlled or if it is just a fixed gain PA?

RadioMasterRC, 2024-03-28:
> Hmm, the SX1276 supports +20dBm and the LR1121 +22 dBm output. Unless there is an external PA it is hard to see how +30dBm could be achieved. Any information if there is an external PA and how it is controlled or if it is just a fixed gain PA?

Hi @dhandzic The Bandit Nano uses a Skyworks SKY66122 PA. The PA Voltage is controlled via the ESP32 DAC port and works together with the SX1276 RF power to set the output power.

dhandzic, 2024-03-28:
> > Hmm, the SX1276 supports +20dBm and the LR1121 +22 dBm output. Unless there is an external PA it is hard to see how +30dBm could be achieved. Any information if there is an external PA and how it is controlled or if it is just a fixed gain PA?
> 
> Hi @dhandzic The Bandit Nano uses a Skyworks SKY66122 PA. The PA Voltage is controlled via the ESP32 DAC port and works together with the SX1276 RF power to set the output power. The effective power control range is 100mw - 1000mw
> 
> Additional info: https://github.com/ExpressLRS/ExpressLRS/tree/master/src/lib/POWERMGNT
> 
> The Bandit Nano uses the "DACWRITE" function.

Thank you for the information. Cool with a +30 dBm node as a repeater. Do you have any figures on the standby current consumption?

dhandzic, 2024-03-28:
> Hi @dhandzic The Bandit Nano uses a Skyworks SKY66122 PA. The PA Voltage is controlled via the ESP32 DAC port and works together with the SX1276 RF power to set the output power. The effective power [...]

Could the device be used for 869 MHz too or is the RX filter tuned for 915 MHz?

mhVuF, 2024-03-28:
> Thank you for the information. Cool with a +30 dBm node as a repeater. Do you have any figures on the standby current consumption?

From the top oh my head, I measured about 2.3W on ELRS while transmitting 30.1dbm, and around 60mA while idle. I'll edit this post later to add more details.

thebentern, 2024-04-14:
If somebody wants to add official support for this, we can for sure integrate it, but I'm closing this as an active issue. I don't have the bandwidth to integrate more SX1276 based boards with all of the backlog of devices to support.

closed 2024-04-14 by thebentern

uberhalit, 2024-04-14:
> If somebody wants to add official support for this, we can for sure integrate it, but I'm closing this as an active issue. I don't have the bandwidth to integrate more SX1276 based boards with all of the backlog of devices to support.

Is this subject open to discussion? As far as I am aware, this device should have high priority, as it is the only portable 900MHz device with adequate output power for dense urban environments. I've noticed a lot of interest lately, and the top complaint in YT comments is the limited range without them even realizing that their device is limited to 125mW (I guess the 30dbm max output setting is a bit misleading). It would be unfortunate if potential users were to abandon the project because they were unable to obtain a proper link in complex environments.

gjelsoe, 2024-04-29:
> > If somebody wants to add official support for this, we can for sure integrate it, but I'm closing this as an active issue. I don't have the bandwidth to integrate more SX1276 based boards with all of the backlog of devices to support.
> 
> Is this subject open to discussion? As far as I am aware, this device should have high priority, as it is the only portable 900MHz device with adequate output power for dense urban environments. I've noticed a lot of interest lately, and the top complaint in YT comments is the poor range in cities without them realizing that their device is limited to 125mW (I guess the 30dbm max output setting is a bit misleading). It would be unfortunate if potential users were to abandon the project because they were unable to obtain a proper link in complex environments.

Are you still working on this?

thebentern, 2024-04-29:
> it is the only portable 900MHz device with adequate output power for dense urban environments

The G2 station can achieve more power than this radio, and has an SX1262 which features a much more sensitive RX end, which makes a greater difference than raw transmit power in my experience, with multiple A/B tests of SX1262 with RX boosted gain w/ 22dbm TX vs 30dbm setups. I'm afraid 30dbm will not be the silver bullet expectations would have you believe. This is a line of sight technology.

KingRikkie, 2024-04-29:
> The G2 station can achieve more power than this radio, and has an SX1262 which features a much more sensitive RX end, which makes a greater difference than raw transmit power in my experience, with [...]

Sorry for squeezing myself into this convo but the G2 "stationary" is not really a comparable portable device. It's 3 times the size, more than double the price (before taxes) and currently unavailable. I much prefer to use existing ELRS hardware that serves multiple purposes.

thebentern, 2024-04-29:
More expensive, I'll grant you. But it is a quite portable device.  
[image]

KingRikkie, 2024-05-01:
> More expensive, I'll grant you. But it is a quite portable device. 

Although still portable, this kind of confirms how much bigger it is compared to the Bandit. I am unaware of your involvement in the G2 project or its vendor, however, the cost is beyond reasonable to me. Including import fees and shipping, this comes up to 150€ + another 10€ for an 868MHz antenna. My local FPV shop has the Bandit on sale for 42€, so I could buy four Bandits for the price of one G2. I'm sorry, but this isn't in any way an alternative.

uberhalit, 2024-05-02:
Regardless of dimensions and price, I think we can all agree that there is a profound lack of higher power devices. Moreover, it is unquestionable that the majority of users would prefer to utilize a device they might already own rather than buying a costly new one. 

> Are you still working on this?
Unfortunately my university schedule is currently too busy to continue. Feel free to pick it up.

gjelsoe, 2024-05-03:
> > Are you still working on this?
> 
> Unfortunately my university schedule is currently too busy to continue. Feel free to pick it up.

Tried too, as [RadioMasterRC](https://github.com/RadioMasterRC) wrote but nothing happened.

spcano01, 2024-05-05:
Larger, unavailable, 4X more expensive, and not available mainstream (single consulting group in China - nice guys though!) 

It would be great to have this as an option, as I can easily see this as the easiest/cheapest/most-consumer-friendly (like T-Echo) device available.

garthvh, 2024-05-05:
> Larger, unavailable, 4X more expensive, and not available mainstream (single consulting group in China - nice guys though!)
> 
> It would be great to have this as an option, as I can easily see this as the easiest/cheapest/most-consumer-friendly (like T-Echo) device available.

Nothing is stopping you from giving it a try, a 1276 with a fan is not a target any core contributors are very interested in.