meshtastic/firmware #3498: [Feature Request]: Add AHT20 Sensor

b8b8, 2024-03-27:
### Platform

NRF52, ESP32

### Description

Can the AHT20+BMP280 sensor be added to support. Currently just the humidity is NOT shown. Everything else seems to work.

https://www.aliexpress.com/item/1005006288395566.html

jaredquinn, 2024-04-06:
> ### Description
> Can the AHT20+BMP280 sensor be added to support. Currently just the humidity is NOT shown. Everything else seems to work.

The AHT20 does not currently seem to be supported; and the BMP280 is providing your pressure and temperature readings (the BMP280 does not do humidity).

I'll take a look at adding support this weekend, I should have an AHT20 laying around somewhere, if not I may need your assistance in testing.

b8b8, 2024-04-06:
sure no problem, and honestly the only reason i am asking is i bought 3x of these not realizing. If it end up being hassle they are easy to use with other projects.

jaredquinn, 2024-04-06:
> sure no problem, and honestly the only reason i am asking is i bought 3x of these not realizing. If it end up being hassle they are easy to use with other projects.

What is your skill level?  Are you able to build and flash from command line?
I have added AHT10 & AHT20 support in my fork.

I need to get some more info from other devs as if just supporting the AHT side of this will be enough combined with the existing BMP support; or if I should add another Telemetry Sensor that uses both.

caveman99, 2024-04-07:
Support looks fine with me, it's just we dont have room for more than one temperature in the telemetrypackets, if you have 2 of them active, it's a bit of a gamble which one was the latest to update :-) - also you'd need a separate PR gaianst the protobuf repo to support a new sensor.

jaredquinn, 2024-04-07:
> It's a bit of a gamble which one was the latest to update :-) 

Perhaps if both a BMP280 and AHT20 are detected we only take the pressure readings from the AHT (this seems to be the suggested use case and the way the wording on product pages implies)?

I will hold off until I can test it properly before I raise a PR - unfortunately I didn't have an AHT20 handy (I had 9 BMP280s handy); so I've ordered a bunch of the board @b8b8 mentioned.

b8b8, 2024-04-07:
> > sure no problem, and honestly the only reason i am asking is i bought 3x of these not realizing. If it end up being hassle they are easy to use with other projects.
> 
> What is your skill level? Are you able to build and flash from command line? I have added AHT10 & AHT20 support in my fork at https://github.com/jaredquinn/meshtastic-firmware/tree/add-aht10-telem-sensor
> 
> I need to get some more info from other devs as if just supporting the AHT side of this will be enough combined with the existing BMP support; or if I should add another Telemetry Sensor that uses both.

Lets say skill level is low. I havent actually done proper programming since University 10 years ago and am a mechanical engineer in my day job.

jaredquinn, 2024-04-08:
> Lets say skill level is low. I havent actually done proper programming since University 10 years ago and am a mechanical engineer in my day job.

No programming required here; just a build from source and flashing as per the documentation.   While I wasn't able to find any AHT20s in my kit I have ordered some of the board you mentioned;  I am happy to wait the estimated week for their arrival.

quimnut, 2024-07-01:
I get temp, humidity and pressure on my aht20+bmp280 module.

b8b8, 2024-07-01:
yup same with me. everything is working well.

closed 2024-09-16 by fifieldt

b8b8, 2024-10-14:
This has stopped displaying humidity recently.

quimnut, 2024-10-14:
> I get temp, humidity and pressure on my aht20+bmp280 module.

Yes, it's gone in a recent update for me also.