meshtastic/firmware #3501: [Bug]: Lora T3S3 sleeps immediately once router mode is enabled.

shleeable, 2024-03-27:
### Category

Hardware Compatibility

### Hardware

T-Beam S3

### Firmware Version

2.2.24

### Description

Detail: I've added a battery and plugged in USB to a dedicated powerbrick to power the router

Replication: On a fresh install of my two new T3S3s. I've configured all of the setting to be correct using the web via serial and iOS via bluetooth.

Once I set the unit from client mode to router mode. It "bricks" itself by sleeping immediately after booting.
This drops the bluetooth and usb serial connection.... clicking a button triggers the resuming screen, and it boots, then immediately sleeps again.

I've replicated this multiple times and it is the client to router step which seems to trigger bad defaults regarding power/sleeping.

This is a router, I never want it to sleep while it has any power via battery/USB.

Holyszewski, 2024-03-28:
This is the correct router behavior in my opinion and in accordance with https://meshtastic.org/docs/configuration/radio/device/ .
In the ROUTER role, only the CPU is asleep, the LoRa transceiver is active and when a message arrives the device wakes up, resolves and broadcasts what is needed and goes back to sleep.

thebentern, 2024-03-28:
If you don't want the device to sleep, I recommend using the ROUTER_CLIENT role.

closed 2024-03-28 by thebentern

shleeable, 2024-03-30:
If this is expected behaviour, why doesn't it match the behaviour from other devices like the RAK?

That's running router mode without the autosleep.. How can I meant to administrator a node if it auto sleeps?