meshtastic/firmware #3503: [Bug]: Administration via admin channel does not work

Holyszewski, 2024-03-28:
### Category

Other

### Hardware

Heltec V3

### Firmware Version

2.3.2.63df972

### Description

Device with FW 2.3.2 cannot be controlled via admin channel for some reason.
Tested on Heltec v3 and Android app version 2.2.23.
When I use an identical admin channel key but install an older FW version on the device, e.g. 2.2.23, the administration works normally. 

The FW version of the node used for control does not matter, only the FW version of the target remote node matters.

thebentern, 2024-03-28:
Could you collect some serial logs to help debug this issue?

Holyszewski, 2024-03-28:
Serial log from the device connected to phone:
```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbe4
load:0x403cc700,len:0x2a38
entry 0x403c98d4
E (349) esp_core_dump_flash: No core dump partition found!
E (349) esp_core_dump_flash: No core dump partition found!
”Ă @INFO  | ??:??:?? 0 

//\ E S H T /\ S T / C

INFO  | ??:??:?? 0 Booted, wake cause 0 (boot count 1), reset_reason=reset
DEBUG | ??:??:?? 0 Filesystem files (495616/1048576 Bytes):
DEBUG | ??:??:?? 0  /prefs/channels.proto (98 Bytes)
DEBUG | ??:??:?? 0  /prefs/config.proto (122 Bytes)
DEBUG | ??:??:?? 0  /prefs/db.proto (469 Bytes)
DEBUG | ??:??:?? 0  /prefs/module.proto (112 Bytes)
DEBUG | ??:??:?? 0  /static/.gitkeep (0 Bytes)
DEBUG | ??:??:?? 0  /static/Logo_Black.svg.gz (602 Bytes)
DEBUG | ??:??:?? 0  /static/Logo_White.svg.gz (610 Bytes)
DEBUG | ??:??:?? 0  /static/apple-touch-icon.png.gz (3164 Bytes)
DEBUG | ??:??:?? 0  /static/favicon.ico.gz (2270 Bytes)
DEBUG | ??:??:?? 0  /static/icon.svg.gz (852 Bytes)
DEBUG | ??:??:?? 0  /static/index-CGqDWSD_.js.gz (348 Bytes)
DEBUG | ??:??:?? 0  /static/index-D5d-WaNC.css.gz (16236 Bytes)
DEBUG | ??:??:?? 0  /static/index-nsbd6Oaw.js.gz (205708 Bytes)
DEBUG | ??:??:?? 0  /static/index.html.gz (547 Bytes)
DEBUG | ??:??:?? 0  /static/maplibre-gl-BSf4l8UU.js.gz (210198 Bytes)
DEBUG | ??:??:?? 0  /static/robots.txt.gz (42 Bytes)
DEBUG | ??:??:?? 0  /static/site.webmanifest.gz (197 Bytes)
[   695][I][esp32-hal-i2c.c:75] i2cInit(): Initialising I2C Master: sda=41 scl=42 freq=100000
[   696][I][esp32-hal-i2c.c:75] i2cInit(): Initialising I2C Master: sda=17 scl=18 freq=100000
DEBUG | ??:??:?? 0 Using analog input 1 for battery level
INFO  | ??:??:?? 0 ADCmod: ADC Characterization based on Two Point values and fitting curve coefficients stored in eFuse
INFO  | ??:??:?? 0 Scanning for i2c devices...
[   729][W][Wire.cpp:301] begin(): Bus already started in Master Mode.
DEBUG | ??:??:?? 0 Scanning for i2c devices on port 2
[   758][W][Wire.cpp:301] begin(): Bus already started in Master Mode.
DEBUG | ??:??:?? 0 Scanning for i2c devices on port 1
DEBUG | ??:??:?? 0 I2C device found at address 0x3c
INFO  | ??:??:?? 0 ssd1306 display found
INFO  | ??:??:?? 0 ssd1306 display found
DEBUG | ??:??:?? 0 0x3 subtype probed in 2 tries 
INFO  | ??:??:?? 0 1 I2C devices found
DEBUG | ??:??:?? 0 acc_info = 0
INFO  | ??:??:?? 0 Meshtastic hwvendor=43, swver=2.3.2.63df972
DEBUG | ??:??:?? 0 Setting random seed 3238737436
DEBUG | ??:??:?? 0 Total heap: 294240
DEBUG | ??:??:?? 0 Free heap: 257744
DEBUG | ??:??:?? 0 Total PSRAM: 0
DEBUG | ??:??:?? 0 Free PSRAM: 0
DEBUG | ??:??:?? 0 NVS: UsedEntries 86, FreeEntries 544, AllEntries 630, NameSpaces 3
DEBUG | ??:??:?? 0 Setup Preferences in Flash Storage
DEBUG | ??:??:?? 0 Number of Device Reboots: 11
DEBUG | ??:??:?? 0 OTA firmware version 0.2.1.33451a5
INFO  | ??:??:?? 0 Initializing NodeDB
INFO  | ??:??:?? 0 Loading /prefs/db.proto
INFO  | ??:??:?? 0 Loaded saved devicestate version 22, with nodecount: 2
INFO  | ??:??:?? 0 Loading /prefs/config.proto
INFO  | ??:??:?? 0 Loaded saved config version 22
INFO  | ??:??:?? 0 Loading /prefs/module.proto
INFO  | ??:??:?? 0 Loaded saved moduleConfig version 22
INFO  | ??:??:?? 0 Loading /prefs/channels.proto
INFO  | ??:??:?? 0 Loaded saved channelFile version 22
[... 703 lines cut ...]
DEBUG | ??:??:?? 46 [DeviceTelemetryModule] ESP32 crypt fr=da576470, num=721eed56, numBytes=28!
DEBUG | ??:??:?? 46 [DeviceTelemetryModule] enqueuing for send (id=0x721eed56 fr=0x70 to=0xff, WantAck=0, HopLim=3 Ch=0x8 encrypted hopStart=3 priority=10)
DEBUG | ??:??:?? 46 [DeviceTelemetryModule] txGood=2,rxGood=3,rxBad=0
DEBUG | ??:??:?? 46 [DeviceTelemetryModule] Using channel 0 (hash 0x8)
DEBUG | ??:??:?? 46 [DeviceTelemetryModule] Expanding short PSK #1
DEBUG | ??:??:?? 46 [DeviceTelemetryModule] Using AES128 key!
DEBUG | ??:??:?? 46 [DeviceTelemetryModule] ESP32 crypt fr=da576470, num=721eed56, numBytes=28!
DEBUG | ??:??:?? 46 [DeviceTelemetryModule] decoded message (id=0x721eed56 fr=0x70 to=0xff, WantAck=0, HopLim=3 Ch=0x0 Portnum=67 hopStart=3 priority=10)
DEBUG | ??:??:?? 46 [RadioIf] Starting low level send (id=0x721eed56 fr=0x70 to=0xff, WantAck=0, HopLim=3 Ch=0x8 encrypted hopStart=3 priority=10)
DEBUG | ??:??:?? 46 [RadioIf] (bw=250, sf=11, cr=4/5) packet symLen=8 ms, payloadSize=44, time 559 ms
DEBUG | ??:??:?? 46 [RadioIf] AirTime - Packet transmitted : 559ms
DEBUG | ??:??:?? 46 [RadioIf] Completed sending (id=0x721eed56 fr=0x70 to=0xff, WantAck=0, HopLim=3 Ch=0x8 encrypted hopStart=3 priority=10)
DEBUG | ??:??:?? 49 [RadioIf] (bw=250, sf=11, cr=4/5) packet symLen=8 ms, payloadSize=44, time 559 ms
DEBUG | ??:??:?? 49 [RadioIf] Lora RX (id=0x721eed56 fr=0x70 to=0xff, WantAck=0, HopLim=2 Ch=0x8 encrypted rxSNR=6.5 rxRSSI=-20 hopStart=3)
DEBUG | ??:??:?? 49 [RadioIf] AirTime - Packet received : 559ms
DEBUG | ??:??:?? 49 [Router] Rx someone rebroadcasting for us (id=0x721eed56 fr=0x70 to=0xff, WantAck=0, HopLim=2 Ch=0x8 encrypted rxSNR=6.5 rxRSSI=-20 hopStart=3)
DEBUG | ??:??:?? 49 [Router] didn't find pending packet
DEBUG | ??:??:?? 49 [Router] Found existing packet record for fr=0xda576470,to=0xffffffff,id=0x721eed56
DEBUG | ??:??:?? 49 [Router] Found existing packet record for fr=0xda576470,to=0xffffffff,id=0x721eed56
DEBUG | ??:??:?? 49 [Router] Add packet record (id=0x721eed56 fr=0x70 to=0xff, WantAck=0, HopLim=2 Ch=0x8 encrypted rxSNR=6.5 rxRSSI=-20 hopStart=3)
DEBUG | ??:??:?? 49 [Router] Ignoring incoming msg, because we've already seen it (id=0x721eed56 fr=0x70 to=0xff, WantAck=0, HopLim=2 Ch=0x8 encrypted rxSNR=6.5 rxRSSI=-20 hopStart=3)
DEBUG | ??:??:?? 49 [Router] cancelSending id=0x721eed56, removed=0
DEBUG | ??:??:?? 49 [Router] Incoming message was filtered 0xda576470
DEBUG | ??:??:?? 61 [Power] Battery: usbPower=0, isCharging=0, batMv=3915, batPct=72
```

GUVWAF, 2024-03-28:
Somebody on Discord mentioned that updating the Android app to >=2.3.0 solved the issue for them.

Might be related to https://github.com/meshtastic/Meshtastic-Android/issues/933.

Mictronics, 2024-03-29:
That's related to the Android app being 2.2.23 and firmware 2.3.x.
I had the same problem. You may experience also that traceroute to the 2.3.x nodes doesn't work anymore.
Update the Android app to a 2.3.x version.