meshtastic/firmware #3511: [Bug]: RPI3B 64bit ARM - web server error on start up

cohen-sydney, 2024-03-30:
### Category

Other

### Hardware

Other

### Firmware Version

2.3.2.63df972

### Description

I have installed on RPI3B 64bit Bookworm.
on running command "meshtasticd"

get the following error 

INFO  | ??:??:?? 0 Webserver started ....
ERROR | ??:??:?? 0 Error starting Web Server framework, error number: 4

any help??

### Relevant log output

```Shell
meshtasticd
Portduino is starting, HWID=1, VFS root at /home/andrew/.portduino/default
Setting up Meshtastic on Portduino...
Using /etc/meshtasticd/config.yaml as config file
�@INFO  | ??:??:?? 0

//\ E S H T /\ S T / C

DEBUG | ??:??:?? 0 Filesystem files:
DEBUG | ??:??:?? 0  /prefs/.. (0 Bytes)
DEBUG | ??:??:?? 0  /prefs/. (0 Bytes)
DEBUG | ??:??:?? 0  /prefs/db.proto (512 Bytes)
DEBUG | ??:??:?? 0  /prefs/channels.proto (57 Bytes)
DEBUG | ??:??:?? 0  /.. (0 Bytes)
DEBUG | ??:??:?? 0  /. (0 Bytes)
INFO  | ??:??:?? 0 No I2C device configured, skipping.
INFO  | ??:??:?? 0 Scanning for i2c devices...
INFO  | ??:??:?? 0 No I2C devices found
INFO  | ??:??:?? 0 Meshtastic hwvendor=37, swver=2.3.2.63df972
INFO  | ??:??:?? 0 Initializing NodeDB
INFO  | ??:??:?? 0 Loading /prefs/db.proto
INFO  | ??:??:?? 0 Loaded saved devicestate version 22, with nodecount: 1
/home/andrew/.portduino/default/prefs/config.proto does not exist
INFO  | ??:??:?? 0 No /prefs/config.proto preferences found
INFO  | ??:??:?? 0 Installing default LocalConfig
INFO  | ??:??:?? 0 Setting default channel and radio preferences!
DEBUG | ??:??:?? 0 Expanding short PSK #1
INFO  | ??:??:?? 0 Wanted region 0, using UNSET
/home/andrew/.portduino/default/prefs/module.proto does not exist
INFO  | ??:??:?? 0 No /prefs/module.proto preferences found
INFO  | ??:??:?? 0 Installing default ModuleConfig
INFO  | ??:??:?? 0 Loading /prefs/channels.proto
INFO  | ??:??:?? 0 Loaded saved channelFile version 22
/home/andrew/.portduino/default/oem/oem.proto does not exist
INFO  | ??:??:?? 0 No /oem/oem.proto preferences found
DEBUG | ??:??:?? 0 cleanupMeshDB purged 0 entries
DEBUG | ??:??:?? 0 Using nodenum 0xeb4db26b
DEBUG | ??:??:?? 0 Expanding short PSK #1
INFO  | ??:??:?? 0 Wanted region 0, using UNSET
DEBUG | ??:??:?? 0 Read RTC time as 1711771246
DEBUG | ??:??:?? 0 Running without GPS.
DEBUG | ??:??:?? 0 NeighborInfoModule is disabled
/home/andrew/.portduino/default/prefs/cannedConf.proto does not exist
INFO  | ??:??:?? 0 No /prefs/cannedConf.proto preferences found
INFO  | ??:??:?? 0 CannedMessageModule is enabled
DEBUG | ??:??:?? 0 Attempting to activate sx1262 radio on SPI port /dev/spidev0.0
DEBUG | ??:??:?? 0 SX126xInterface(cs=21, irq=16, rst=18, busy=20)
DEBUG | ??:??:?? 0 SX126X_DIO3_TCXO_VOLTAGE not defined, not using DIO3 as TCXO reference voltage
INFO  | ??:??:?? 0 Starting meshradio init...
DEBUG | ??:??:?? 0 (bw=250, sf=11, cr=4/5) packet symLen=8 ms, payloadSize=0, time 231 ms
DEBUG | ??:??:?? 0 (bw=250, sf=11, cr=4/5) packet symLen=8 ms, payloadSize=253, time 2115 ms
INFO  | ??:??:?? 0 Radio freq=906.875, config.lora.frequency_offset=0.000
INFO  | ??:??:?? 0 Set radio: region=UNSET, name=LongFast, config=0, ch=19, power=30
INFO  | ??:??:?? 0 Radio myRegion->freqStart -> myRegion->freqEnd: 902.000000 -> 928.000000 (26.000000 mhz)
INFO  | ??:??:?? 0 Radio myRegion->numChannels: 104 x 250.000kHz
INFO  | ??:??:?? 0 Radio channel_num: 20
INFO  | ??:??:?? 0 Radio frequency: 906.875000
INFO  | ??:??:?? 0 Slot time: 42 msec
INFO  | ??:??:?? 0 Set radio: final power level=22
INFO  | ??:??:?? 0 SX126x init result 0
INFO  | ??:??:?? 0 Frequency set to 906.875000
INFO  | ??:??:?? 0 Bandwidth set to 250.000000
INFO  | ??:??:?? 0 Power output set to 22
DEBUG | ??:??:?? 0 Current limit set to 140.000000
DEBUG | ??:??:?? 0 Current limit set result 0
DEBUG | ??:??:?? 0 Setting DIO2 as RF switch
DEBUG | ??:??:?? 0 Using MCU pin -1 as RXEN and pin -1 as TXEN to control RF switching
INFO  | ??:??:?? 0 Set RX gain to boosted mode; result: 0
INFO  | ??:??:?? 0 SX1262 Radio init succeeded, using SX1262 radio
INFO  | ??:??:?? 0 Using webserver port from yaml config. 443
INFO  | ??:??:?? 0 Webserver started ....
ERROR | ??:??:?? 0 Error starting Web Server framework, error number: 4
INFO  | ??:??:?? 0 API server listening on TCP port 4403
DEBUG | ??:??:?? 0 (bw=250, sf=11, cr=4/5) packet symLen=8 ms, payloadSize=237, time 2000 ms
DEBUG | ??:??:?? 0 LoRA bitrate = 118.500000 bytes / sec
INFO  | ??:??:?? 0 PowerFSM init, USB power=1
DEBUG | ??:??:?? 0 Enter state: BOOT
INFO  | ??:??:?? 30 [NodeInfoModule] Sending our nodeinfo to mesh (wantReplies=1)
INFO  | ??:??:?? 30 [NodeInfoModule] sending owner !eb4db26b/Meshtastic b26b/b26b
DEBUG | ??:??:?? 30 [NodeInfoModule] Initial packet id 1804289383, numPacketId 4294967295
DEBUG | ??:??:?? 30 [NodeInfoModule] Update DB node 0xeb4db26b, rx_time=0
DEBUG | ??:??:?? 30 [NodeInfoModule] handleReceived(LOCAL) (id=0x6b8b4569 fr=0x6b to=0xff, WantAck=0, HopLim=3 Ch=0x0 Portnum=4 WANTRESP priority=10)
DEBUG | ??:??:?? 30 [NodeInfoModule] No modules interested in portnum=4, src=LOCAL
DEBUG | ??:??:?? 30 [NodeInfoModule] localSend to channel 0
DEBUG | ??:??:?? 30 [NodeInfoModule] Add packet record (id=0x6b8b4569 fr=0x6b to=0xff, WantAck=0, HopLim=3 Ch=0x0 Portnum=4 WANTRESP priority=10)
DEBUG | ??:??:?? 30 [NodeInfoModule] Expanding short PSK #1
DEBUG | ??:??:?? 30 [NodeInfoModule] Using AES128 key!
DEBUG | ??:??:?? 30 [NodeInfoModule] Installing AES128 key!
WARN  | ??:??:?? 30 [NodeInfoModule] send - lora tx disable because RegionCode_Unset
```

jp-bennett, 2024-03-30:
Try installing the libulfius2.7 package, if it's not already installed.

cohen-sydney, 2024-03-30:
I changed the Port to 8443 and the error went away, just trying to connect via Wifi network.

jp-bennett, 2024-03-30:
> I changed the Port to 8443 and the error went away, just trying to connect via Wifi network.

Ah, if you're running the command as a non-root user, it probably doesn't have permission to claim port 443. Or maybe something else already has that port claimed. Closing for now, re-open if this pops back up!

closed 2024-03-30 by jp-bennett

cohen-sydney, 2024-03-30:
how do you run this at startup?

jp-bennett, 2024-03-30:
> how do you run this at startup? like headless??

The .Deb installs a systemd service file. So "systemctl enable meshtasticd" will enable it at boot.

eric-becker, 2024-04-12:
I had the exact same error with another service listening on port 443. Can you verify if a process is listening on 443 (or whatever port you have configured in your /etc/meshtasticd/config.yaml?

`mesh@rpi-mesh:~ $ sudo lsof -i :https
COMMAND  PID USER   FD   TYPE DEVICE SIZE/OFF NODE NAME
nc      1524 root    3u  IPv6  38842      0t0  TCP *:https (LISTEN)
nc      1524 root    4u  IPv4  38843      0t0  TCP *:https (LISTEN)`

`mesh@rpi-mesh:~ $ sudo netstat -ltnp | grep -w ':443'
tcp6       0      0 :::443                  :::*                    LISTEN      1524/nc `