meshtastic/firmware #3512: [Bug]:Power Save mode no longer letting screen to wake

5kamisama5, 2024-03-30:
### Category

Other

### Hardware

Heltec V3

### Firmware Version

2.3.2

### Description

I have a few standalone Heltec V3s with CardKBs that were recently upgraded to firmware 2.3.2. I use them on power save mode. When the screen goes to sleep at 10min I can use the program button to wake it with no issues, but after 15min it no longer wakes the screen when i press the Program button, I can confirm it is still on because the screen wakes up when I send a message to it. It seems something was changed on recent firmwares. I can confirm with another node that has 2.2.22 where it does wake up normally regardless how long the screen is off. 
I've tried to factory reset the node and only turned on Powersave mode and cardkb and it still happens on all nodes with this firmware.

garthvh, 2024-03-30:
The heltec devices may not be the best choice for sleep features

5kamisama5, 2024-03-30:
> The heltec devices may not be the best choice for sleep features

So you suggest i never upgrade past  firmware 2.2.22? Cause it works just fine there.

todd-herbert, 2024-03-30:
Is the device sleeping (LED heartbeat stopped), or is the screen just off?

5kamisama5, 2024-03-30:
> Is the device sleeping (LED heartbeat stopped), or is the screen just off?

LED heartbeat is still working, just the screen is off.

garthvh, 2024-03-30:
> > The heltec devices may not be the best choice for sleep features
> 
> So you suggest i never upgrade past firmware 2.2.22? Cause it works just fine there.

A board with battery detection that is finicky is not going to be a great fit for roles that sleep, sleep requires that the power / battery data be reasonably accurate.

5kamisama5, 2024-03-30:
> battery detection that is finicky is not going to be a great fit for roles that sleep, sleep requires that the power / battery data be reasonably accurate.

it is true it's not the best board but people have it and are using it. A feature that worked before and doesn't work now, means something was changed to break it. I've just taken the time to upgrade to each firmware to pinpoint when it broke. It's not popular among the PROs but people still have this board. 

2.2.22 - Still Works
2.2.23 - Still Works
2.2.24 - Broke Here (Screen not turning on after 15min when button pressed)
2.3.2  - Screen not turning on after 15min
2.3.3  - Screen not turning on after 15min

The device is still has heartbeat, only the screen doesn't turn on when the program button is pressed, screen wakes when I send a message so it is working on power save mode, hope this is enough information to warrant some kind of fix from someone who would take the time to consider it. I am currently starting coding classes so I can be of some use to the community, till then all I can do is point out things I can test with my current ability.

garthvh, 2024-03-30:
Part of why I asked when you were making this recommendation initially is because it should be sleeping like it is now, the behavior you were seeing is not how the is low power works

5kamisama5, 2024-03-30:
I see, so power save mode should be avoided on the V3. Do you know what  settings are overwritten when this is turned on? like screen time/blutooth shut off, etc. I'd like to manually configure them to get as close as possible if I can't use power save mode on this.

garthvh, 2024-03-30:
I don't really understand still what part of is low power is working for the heltec v3, or how much power it is saving.

garthvh, 2024-03-30:
Probably is not Bluetooth or the screen.

5kamisama5, 2024-03-31:
**2.2.23 with power save mode on:**
Normal operation: when screen is on it operates at 109mA
Screen turns off after 10min and it operates at 11mA
When I press the program button after 15min it turns the screen on and goes back to 109mA. This is good.

**2.3.2 with power save mode on:**
Normal operation: when screen is on it operates at 109mA
Screen turns off after 10min and it operates at 105mA
After 15min it goes to 11mA, when I press the program button it doesn't turn on the screen on like before and remains on 11mA. No good.

Power save mode saves 10 times power when on light sleep and that's very desirable on standalone devices, but if I have to restart the device anytime i need to turn on the screen it becomes quite tidious.

[image]
[image]

blueskiesMesh, 2024-03-31:
Yes, I have many sets of Heltec V3 + Cardkb as standalone devices and they are facing the exact same issue.
Can this be resolved in future firmware?

todd-herbert, 2024-03-31:
I *can* reproduce this bug on Heltec Wireless Paper. 
(Although first requires a hotfix to allow light-sleep, https://github.com/meshtastic/firmware/issues/3517)

Not sure if I personally can help, but I'll take a look at it.

brunette-mod, 2024-04-01:
Is there a solution for this problem? Asking because I have a set of V3's in use as standalone comms for a group of us who use them for hiking and camping. I had been good about updating the firmware on all of them. Having the screen come back to wake mode (when triggered by the program button) is the behavior that people were used to. Also because we put the devices in backpacks etc then look at them periodically to check for messages. TIA

todd-herbert, 2024-04-01:
It should hopefully be fixed with pull request https://github.com/meshtastic/firmware/pull/3521

brunette-mod, 2024-04-01:
@todd-herbert  Thank you so much! Err, need to ask this, but I do have some coding experience, been wondering which subjects I'd need to get up to speed on in order to contribute to the project. Any pointers? Thank you. :)

todd-herbert, 2024-04-01:
@brunette-mod Hey I'm not nearly in-the-loop enough to give you an answer there, but the devs are very active and very helpful over on the [Meshtastic Discord Server](https://discord.com/invite/ktMAKGBnBs).

closed 2024-04-09 by thebentern