meshtastic/firmware #3520: [Bug]:  (SX1262) SX126x standby failed with error -707

el-samiyel, 2024-03-31:
### Category

Hardware Compatibility

### Hardware

DIY

### Firmware Version

2.2.23

### Description

SX126x standby failed with error -707

Can anyone help me to understand why I am getting this?

I am using a E22 SX1262 chip, tried both configurations (ESP32 controls TXEN/RXEN and auto control) but I get the same result.

### Relevant log output

```Shell
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13192
load:0x40080400,len:3028
entry 0x400805e4
E (789) esp_core_dump_flash: No core dump
```

jp-bennett, 2024-03-31:
That error almost always means a problem with the control of the oscillator. Take a look at SX126X_DIO3_TCXO_VOLTAGE in other variant files.

el-samiyel, 2024-03-31:
> That error almost always means a problem with the control of the oscillator. Take a look at SX126X_DIO3_TCXO_VOLTAGE in other variant files.

Thanks, I've had a look and its set like the others to be honest '#define SX126X_DIO3_TCXO_VOLTAGE 1.8', not entirely sure what this references and these no mention of it in the data sheet. Any other suggestions?

jp-bennett, 2024-03-31:
> Thanks, I've had a look and its set like the others to be honest '#define SX126X_DIO3_TCXO_VOLTAGE 1.8', not entirely sure what this references and these no mention of it in the data sheet. Any [...]

In that case, try leaving that line out. Source: https://jgromes.github.io/RadioLib/group__status__codes.html#gabc695a4fae689e856ae6f618e334066f

el-samiyel, 2024-03-31:
> In that case, try leaving that line out. Source: https://jgromes.github.io/RadioLib/group__status__codes.html#gabc695a4fae689e856ae6f618e334066f

Just gave that a try, but the result was SX1262 not found. When I changed back to 1.8v it finds the sx1262 again but I get into a reboot loop when trying to set the region.

`SX126x init result -707`
`SX126x standby failed with error -707`

closed 2024-04-02 by el-samiyel