meshtastic/firmware #3528: [Bug]: JSON Over MQTT will send accept packets

degan6, 2024-04-01:
### Category

Other

### Hardware

Other

### Firmware Version

2.3.3.8187fa7

### Description

I have been unable to get JSON MQTT to work. I've tired many troubleshooting steps and asked for help on [discord](https://discord.com/channels/867578229534359593/1223450393250500670) but so far nothing has worked.

My setup:

- Confirmed MQTT is working by sending a message and seeing it on MQTT
- Configured a channel named "mqtt" on the Meshtastic device and enabled downlink.
- Ensured MQTT is enabled and JSON mode is active, with encryption turned off.
- Published a message to msh/2/json/mqtt with the following payload:

```
{
    "from": 841446079,
    "type": "sendtext",
    "payload": "klk"
}
```
Received an error in device logs: [mqtt] JSON downlink received on channel not called 'mqtt' or without downlink enabled.

Troubleshooting Steps Taken:

- Verified the channel named "mqtt" exists and downlink is enabled.
- Checked case sensitivity for the channel name.
- Restarted Meshtastic devices and the MQTT broker.
- Confirmed firmware on Meshtastic devices is up to date.
- Verified encryption is turned off to ensure compatibility with MQTT downlink.
- Tried adding channel in the json payload
- Tired many different mqtt topics

Seems to be this line of code
https://github.com/meshtastic/firmware/blob/279464f96d5139920b017d437501233737daf407/src/mqtt/MQTT.cpp#L63

Thanks for your help!

### Relevant log output

```Shell
ortduino is starting, HWID=1, VFS root at /root/.portduino/default
Setting up Meshtastic on Portduino...
Using /etc/meshtasticd/config.yaml as config file
��@INFO  | ??:??:?? 0

//\ E S H T /\ S T / C

INFO  | ??:??:?? 0 No I2C device configured, skipping.
INFO  | ??:??:?? 0 Scanning for i2c devices...
INFO  | ??:??:?? 0 No I2C devices found
INFO  | ??:??:?? 0 Meshtastic hwvendor=37, swver=2.3.3.8187fa7
INFO  | ??:??:?? 0 Initializing NodeDB
INFO  | ??:??:?? 0 Loading /prefs/db.proto
INFO  | ??:??:?? 0 Loaded saved devicestate version 22, with nodecount: 1
INFO  | ??:??:?? 0 Loading /prefs/config.proto
INFO  | ??:??:?? 0 Loaded saved config version 22
INFO  | ??:??:?? 0 Loading /prefs/module.proto
INFO  | ??:??:?? 0 Loaded saved moduleConfig version 22
INFO  | ??:??:?? 0 Loading /prefs/channels.proto
INFO  | ??:??:?? 0 Loaded saved channelFile version 22
/root/.portduino/default/oem/oem.proto does not exist
INFO  | ??:??:?? 0 No /oem/oem.proto preferences found
WARN  | ??:??:?? 0 User disabled encryption
INFO  | ??:??:?? 0 Wanted region 1, using US
Portduino critical error: Failed to open posix file /dev/spidev0.0, errno=2
No hardware spi chip found...
/root/.portduino/default/prefs/cannedConf.proto does not exist
INFO  | ??:??:?? 0 No /prefs/cannedConf.proto preferences found
INFO  | ??:??:?? 0 CannedMessageModule is enabled
INFO  | ??:??:?? 0 Starting meshradio init...
INFO  | ??:??:?? 0 Radio freq=919.625, config.lora.frequency_offset=0.000
INFO  | ??:??:?? 0 Set radio: region=US, name=cam, config=0, ch=70, power=30
INFO  | ??:??:?? 0 Radio myRegion->freqStart -> myRegion->freqEnd: 902.000000 -> 928.000000 (26.000000 mhz)
INFO  | ??:??:?? 0 Radio myRegion->numChannels: 104 x 250.000kHz
INFO  | ??:??:?? 0 Radio channel_num: 71
INFO  | ??:??:?? 0 Radio frequency: 919.625000
INFO  | ??:??:?? 0 Slot time: 42 msec
INFO  | ??:??:?? 0 Using SIMULATED radio!
INFO  | ??:??:?? 0 Using webserver port from yaml config. 443
INFO  | ??:??:?? 0 Webserver started ....
INFO  | ??:??:?? 0 Web Server framework started on port: 443
INFO  | ??:??:?? 0 Web Server root /usr/share/doc/meshtasticd/web
INFO  | ??:??:?? 0 API server listening on TCP port 4403
INFO  | ??:??:?? 0 PowerFSM init, USB power=1
INFO  | ??:??:?? 0 [mqtt] Attempting to connect directly to MQTT server localhost, port: 1883, username: mesh-pi, password: o5l169+D+e#Q
INFO  | ??:??:?? 0 [mqtt] MQTT connected
INFO  | ??:??:?? 0 [mqtt] published online=1
INFO  | ??:??:?? 0 [mqtt] Subscribing to msh/2/e/LongFast/#
INFO  | ??:??:?? 0 [mqtt] Subscribing to msh/2/json/LongFast/#
INFO  | ??:??:?? 0 [mqtt] Subscribing to msh/2/e/mqtt/#
INFO  | ??:??:?? 0 [mqtt] Subscribing to msh/2/json/mqtt/#
INFO  | ??:??:?? 0 [mqtt] Subscribing to msh/2/e/cam/#
INFO  | ??:??:?? 0 [mqtt] Subscribing to msh/2/json/cam/#
INFO  | ??:??:?? 6 Client wants config, nonce=105912007
INFO  | ??:??:?? 6 Starting API client config
INFO  | ??:??:?? 6 getFromRadio=STATE_SEND_MY_INFO
INFO  | ??:??:?? 6 getFromRadio=STATE_SEND_METADATA
INFO  | ??:??:?? 6 getFromRadio=STATE_SEND_NODEINFO
INFO  | ??:??:?? 6 nodeinfo: num=0x322772bf, lastseen=0, id=!322772bf, name=AllPi 72bf
INFO  | ??:??:?? 6 getFromRadio=STATE_SEND_NODEINFO
INFO  | ??:??:?? 6 Done sending nodeinfos
INFO  | ??:??:?? 6 getFromRadio=STATE_SEND_CHANNELS
INFO  | ??:??:?? 6 getFromRadio=STATE_SEND_CHANNELS
INFO  | ??:??:?? 6 getFromRadio=STATE_SEND_CHANNELS
INFO  | ??:??:?? 6 getFromRadio=STATE_SEND_CHANNELS
INFO  | ??:??:?? 6 getFromRadio=STATE_SEND_CHANNELS
INFO  | ??:??:?? 6 getFromRadio=STATE_SEND_CHANNELS
INFO  | ??:??:?? 6 getFromRadio=STATE_SEND_CHANNELS
INFO  | ??:??:?? 6 getFromRadio=STATE_SEND_CHANNELS
INFO  | ??:??:?? 6 getFromRadio=STATE_SEND_CONFIG
INFO  | ??:??:?? 6 getFromRadio=STATE_SEND_CONFIG
INFO  | ??:??:?? 6 getFromRadio=STATE_SEND_CONFIG
INFO  | ??:??:?? 6 getFromRadio=STATE_SEND_CONFIG
INFO  | ??:??:?? 6 getFromRadio=STATE_SEND_CONFIG
INFO  | ??:??:?? 6 getFromRadio=STATE_SEND_CONFIG
INFO  | ??:??:?? 6 getFromRadio=STATE_SEND_CONFIG
INFO  | ??:??:?? 6 getFromRadio=STATE_SEND_MODULECONFIG
INFO  | ??:??:?? 6 getFromRadio=STATE_SEND_MODULECONFIG
INFO  | ??:??:?? 6 getFromRadio=STATE_SEND_MODULECONFIG
INFO  | ??:??:?? 6 getFromRadio=STATE_SEND_MODULECONFIG
INFO  | ??:??:?? 6 getFromRadio=STATE_SEND_MODULECONFIG
INFO  | ??:??:?? 6 getFromRadio=STATE_SEND_MODULECONFIG
INFO  | ??:??:?? 7 getFromRadio=STATE_SEND_MODULECONFIG
INFO  | ??:??:?? 7 getFromRadio=STATE_SEND_MODULECONFIG
INFO  | ??:??:?? 7 getFromRadio=STATE_SEND_MODULECONFIG
INFO  | ??:??:?? 7 getFromRadio=STATE_SEND_MODULECONFIG
INFO  | ??:??:?? 7 getFromRadio=STATE_SEND_MODULECONFIG
INFO  | ??:??:?? 7 getFromRadio=STATE_SEND_MODULECONFIG
INFO  | ??:??:?? 7 getFromRadio=STATE_SEND_MODULECONFIG
INFO  | ??:??:?? 7 getFromRadio=STATE_SEND_COMPLETE_ID
INFO  | ??:??:?? 7 getFromRadio=STATE_SEND_PACKETS
WARN  | ??:??:?? 12 [mqtt] JSON downlink received on channel not called 'mqtt' or without downlink enabled.
WARN  | ??:??:?? 13 [mqtt] JSON downlink received on channel not called 'mqtt' or without downlink enabled.
WARN  | ??:??:?? 14 [mqtt] JSON downlink received on channel not called 'mqtt' or without downlink enabled.
WARN  | ??:??:?? 14 [mqtt] JSON downlink received on channel not called 'mqtt' or without downlink enabled.
WARN  | ??:??:?? 15 [mqtt] JSON downlink received on channel not called 'mqtt' or without downlink enabled.
INFO  | ??:??:?? 30 [NodeInfoModule] Sending our nodeinfo to mesh (wantReplies=1)
INFO  | ??:??:?? 30 [NodeInfoModule] sending owner !322772bf/AllPi 72bf/72bf
INFO  | ??:??:?? 30 Telling client we have new packets 2
INFO  | ??:??:?? 30 Telling client we have new packets 4
INFO  | ??:??:?? 30 getFromRadio=STATE_SEND_PACKETS
INFO  | ??:??:?? 30 getFromRadio=STATE_SEND_PACKETS
INFO  | ??:??:?? 30 getFromRadio=STATE_SEND_PACKETS
WARN  | ??:??:?? 62 [mqtt] JSON downlink received on channel not called 'mqtt' or without downlink enabled.
WARN  | ??:??:?? 63 [mqtt] JSON downlink received on channel not called 'mqtt' or without downlink enabled.
WARN  | ??:??:?? 64 [mqtt] JSON downlink received on channel not called 'mqtt' or without downlink enabled.
```

GUVWAF, 2024-04-01:
I see you were using the Linux Native application with simulated radio, so I tried to reproduce it using this as well, with the same payload as you (except for using my own node number), but it works for me:

```
INFO  | ??:??:?? 0 [mqtt] MQTT connected
INFO  | ??:??:?? 0 [mqtt] published online=1
INFO  | ??:??:?? 0 [mqtt] Subscribing to msh/2/e/mqtt/#
INFO  | ??:??:?? 0 [mqtt] Subscribing to msh/2/json/mqtt/#
INFO  | ??:??:?? 28 [mqtt] JSON payload klk, length 3
DEBUG | ??:??:?? 28 [mqtt] Initial packet id 917327548, numPacketId 4294967295
DEBUG | ??:??:?? 28 [mqtt] Update DB node 0xa19cb507, rx_time=0
DEBUG | ??:??:?? 28 [mqtt] handleReceived(LOCAL) (id=0x36ad4ebe fr=0x07 to=0xff, WantAck=0, HopLim=3 Ch=0x0 Portnum=1)
DEBUG | ??:??:?? 28 [mqtt] No modules interested in portnum=1, src=LOCAL
DEBUG | ??:??:?? 28 [mqtt] localSend to channel 0
DEBUG | ??:??:?? 28 [mqtt] Add packet record (id=0x36ad4ebe fr=0x07 to=0xff, WantAck=0, HopLim=3 Ch=0x0 Portnum=1)
DEBUG | ??:??:?? 28 [mqtt] Original length - 3 
DEBUG | ??:??:?? 28 [mqtt] Compressed length - 3 
DEBUG | ??:??:?? 28 [mqtt] Original message - klk 
DEBUG | ??:??:?? 28 [mqtt] Not using compressing message.
DEBUG | ??:??:?? 28 [mqtt] Expanding short PSK #1
DEBUG | ??:??:?? 28 [mqtt] Using AES128 key!
DEBUG | ??:??:?? 28 [mqtt] Installing AES128 key!
DEBUG | ??:??:?? 28 [mqtt] MQTT onSend - Publishing portnum 1 message
DEBUG | ??:??:?? 28 [mqtt] MQTT Publish msh/2/e/mqtt/!00000004, 47 bytes
DEBUG | ??:??:?? 28 [mqtt] got text message of size 3
INFO  | ??:??:?? 28 [mqtt] text message payload is of type plaintext
INFO  | ??:??:?? 28 [mqtt] serialized json message: {"channel":0,"from":2711401735,"hops_away":0,"id":917327550,"payload":{"text":"klk"},"sender":"!00000004","timestamp":0,"to":42949672
INFO  | ??:??:?? 28 [mqtt] JSON publish message to msh/2/json/mqtt/!00000004, 150 bytes: {"channel":0,"from":2711401735,"hops_away":0,"id":917327550,"payload":{"text":"klk"},"sender":"!
DEBUG | ??:??:?? 28 [mqtt] enqueuing for send (id=0x36ad4ebe fr=0x07 to=0xff, WantAck=0, HopLim=3 Ch=0x1e encrypted hopStart=3)
DEBUG | ??:??:?? 28 [mqtt] Set random delay before transmitting.
DEBUG | ??:??:?? 28 [SimRadio] delay done
DEBUG | ??:??:?? 28 [SimRadio] Starting low level send (id=0x36ad4ebe fr=0x07 to=0xff, WantAck=0, HopLim=3 Ch=0x1e encrypted hopStart=3 priority=64)
DEBUG | ??:??:?? 28 [SimRadio] Using channel 0 (hash 0x1e)
DEBUG | ??:??:?? 28 [SimRadio] Expanding short PSK #1
DEBUG | ??:??:?? 28 [SimRadio] Using AES128 key!
DEBUG | ??:??:?? 28 [SimRadio] Installing AES128 key!
DEBUG | ??:??:?? 28 [SimRadio] decoded message (id=0x36ad4ebe fr=0x07 to=0xff, WantAck=0, HopLim=3 Ch=0x0 Portnum=1 hopStart=3 priority=64)
DEBUG | ??:??:?? 28 [SimRadio] (bw=250, sf=8, cr=4/5) packet symLen=1 ms, payloadSize=23, time 60 ms
DEBUG | ??:??:?? 28 [SimRadio] AirTime - Packet transmitted : 60ms
```

I see that you have downlink enabled on multiple channels, so are you sure the `[mqtt] JSON downlink received on channel not called 'mqtt' or without downlink enabled.` log messages occur when you are sending to the MQTT server and not some other node uplinking their JSON messages on another channel?

degan6, 2024-04-01:
Thank for looking into this @GUVWAF, I was able to get it working in the last few minutes. I think this is more a settings / docs issue than anything. I'm going to keep changing settings until I understand the behavior then open a pull on the docs page or another issue.

closed 2024-04-01 by degan6