meshtastic/firmware #3529: 2.2.x and 2.3.2 Nodes Do Not Update Each Other or Traceroute

SenorFusion, 2024-04-01:
I am uncertain if this is an Android issue, but I have not seen iOS users reporting it.

After upgrading nodes and Android devices to 2.3.2 they seem unable to receive updates from 2.2.x nodes

Nodes that I was seeing before and have in my history, now do not update their `Last Seen` time or respond to Traceroutes.

You might could chalk this up to radio conditions, but at least 12 nodes I was seeing all stopped updating, which seems unlikely.

Traceroute from a 2.3.2 Node to a 2.3.2 node = Works as expected

Traceroute from a 2.3.2 Node to 2.2.x nodes = Nothing shown and no "Last seen"

This is similar to meshtastic/Meshtastic-Android#933 but the conclusion there seemed to be that was a 2.2.x issue, and all of my equipment (node and Android) are running 2.3.2

@rcarteraz reported seeing similar things on his nodes.

GUVWAF, 2024-04-01:
I just tried a traceroute from a node on 2.3.3 to one still running 2.2.20 with both version 2.3.1 and 2.3.2 of the Android app, and that works for me. "Last heard" also updates regularly. Tried from a RAK4631 and a Heltec Wireless Paper.

GUVWAF, 2024-04-02:
@rcarteraz  I’m not sure if this is related then, since you also receive packets fine. I’ve seen before the last heard in the node list lags a bit behind.

@SenorFusion Do you see any packets from those nodes in the debug log?
Also, you mentioned it happened after updating to 2.3.2. Can you confirm it works again if you go back to 2.2? Since I cannot reproduce it and 2.3.2 is already beta for a while, I want to rule out other causes first.

SenorFusion, 2024-04-02:
@GUVWAF  I don't have time to reflash back to 2.2 today for testing that but here is a logout output from yesterday.

1. cleared the log
2. Hit traceroute on one of the nodes I used to see but I am not getting updates from - No response shown
3. Grabbed this screenshot
[image]

Is that a response from the node I tried to ping?

I do understand that Traceroute is complicated and doesn't always work for various reasons, but the "Last Updated" is the part that is more concerning. I read on discord somewhere that default update times were changed - maybe there is some issue there?

I will say, I am having a hard time interpreting this or the logs in general, because I don't know how to get the Android app to show the Node Number or User ID, so it's very hard to parse the logs, especially within telemetry of several nodes flowing through.

GUVWAF, 2024-04-02:
No, what is shown here is the Traceroute going out and somebody rebroadcasting it (the `ROUTING_APP` packet). There's no response shown.

The `from` field is the node number of the transmitter and `to` is the one of the destination. In brackets it shows the ID, which is just the hexadecimal representation of the node number. 

The `to` of the traceroute seems formatted weird, but I doubt that's the issue.

Telemetry is coming in every minute from the device you're connected to via your phone.

SenorFusion, 2024-04-02:
> The `from` field is the node number of the transmitter and `to` is the one of the destination. In brackets it shows the ID, which is just the hexadecimal representation of the node number.

Yep make sense - but I don't see that there is any way (on Android) to understand who is who on the Node list vs in the Log.

The node is mounted high up outdoors, so I would have to pull it down to use the CLI or web interface to get more info, but if I pull it down it won't be seeing all the nodes that it is seeing.

GUVWAF, 2024-04-02:
I see, yes I don't think you can see raw node numbers of others in the Android app.

But given that it's high up and the other nodes are far away, might it be something changed when you reinstalled it after updating? E.g. antenna not properly connected anymore, as it seems to not receive anything at all from the nodes further away.

SenorFusion, 2024-04-02:
My node is the same, besides the update, but of course I have no way of knowing what other people have changed.

Looking at things that I can control: I wrote this on Discord but forgot to mention it here.

Node A: one we have been discussing, high up outside.

Node B: On my vehicle, maybe 100ft away from Node A

Since updating both to 2.3.2, I do not see that Node B is online at all unless I manually hit it with a Traceroute from Node A. The Traceroute does work every time though.

Is this expected behavior? Maybe I misunderstand how "Last Seen" is supposed to work.

Does it matter that no phone is connected to Node B? It should still show as online right?

Previously they would both show each other as "Online" all the time and update regularly.

Now the vehicle node just is more and more hours old on the Last Seen unless I manually traceroute it, and the app doesn't seem to count it as an "Online Node" unless I do so.

SenorFusion, 2024-04-02:
I will add that I have never done a ["Factory Erase"](https://meshtastic.org/docs/getting-started/flashing-firmware/nrf52/nrf52-erase/) flash on any of these devices - just a regular flash.

Could that be causing problems?

SenorFusion, 2024-04-02:
> Do you receive any packets from the node in your vehicle? If it's not set to repeater/router role, it should by default send out its device telemetry like battery level every 15 min.

As far as I can tell - no. It responds to a traceroute, but then I receive nothing and Last Seen just gets older and older. It is a regular Client on 2.3.2

> It's hard to debug if you don't control the other far away nodes. There might be someone running 2.3.0 or 2.3.1 that had known issues, or they might have changed channels, etc.

Just as I've been sitting here this afternoon I just tried more testing.

Node F pops up as a new new node. Shows online, battery, all the normal stuff.

Then I get nothing from Node F (as far as I can tell). Last seen gets older and older. Just now, Last Seen was 2+ hours, not online.

I traceroute Node F

Traceroute comes back (1 hop in between)

All the sudden I have Node F online again, Last Seen = Now, battery status etc.

Basically exactly what's happening with my Vehicle node - It's like it's gone until I manually ping it.

Node F is showing Sats connected and ASL - Does that mean its on 2.3.2?

GUVWAF, 2024-04-02:
> Node F is showing Sats connected and ASL - Does that mean its on 2.3.2?

Not necessarily, the data was already there, but it was not yet shown on Android.

SenorFusion, 2024-04-02:
When I have time I will do a "Factory Erase" and Reflash both of my nodes and see if that changes anything.

GUVWAF, 2024-04-03:
As I mentioned on Discord, this can be solved by adding the private channel to node A, because currently it can't decrypt the periodic broadcasts from node B. If you want the firmware to send encrypted packets to the client as well, I think you need to write an RFC because it’s a fundamental change that affects all clients.

That said, in a low-bandwidth mesh I think “online” should not be interpreted like it is on the internet anyway. If a node doesn’t send out any periodic packet, its last heard will not be updated regularly, regardless of encryption. For example: if someone doesn’t need device telemetry, they can decide to increase the broadcast interval. NodeInfo is only sent out every 3 hours by default, and a node with router role has all default broadcast intervals set to 12 hours. We don’t know who is relaying a packet for another node, so it won't update based on that and thus a repeater will never be shown as “online”.

jaredquinn, 2024-04-06:
> I don't exactly know when the Android app considers a node online (maybe heard in last 15 min.?), but "last heard" should update on every received packet. It doesn't matter if it's connected to a phone or not.

2 hours in the firmware; data provided by firmware.

mesh/NodeDB.cpp:766
```
#define NUM_ONLINE_SECS (60 * 60 * 2) // 2 hrs to consider someone offline
```

GUVWAF, 2024-04-06:
> 2 hours in the firmware

Yes, that's indeed in the firmware, but that is not sent to the client.
I believe I found it to be indeed 15 minutes for the Android app:

https://github.com/meshtastic/Meshtastic-Android/blob/e887336da339a6a7de7daa269c39d86e6939b60d/app/src/main/java/com/geeksville/mesh/NodeInfo.kt#L245-L250