meshtastic/firmware #3531: DIY LORA HAT for Raspberry Pi

ON2ON-HAM, 2024-04-01:
### SOC

Other

### Lora IC

Hope RF 96 (SX1278)

### Product Link

http://www.iot4pi.com/en/hardware-out-of-the-limelight/lora-raspberry-pi-gateway-en/

### Description

I use a DIY Lora Hat that was designated as a LORA APRS Igate.
Without any modification to the Hat it is fully usable for MeshtasticD (Linux Native)
[](url)http://www.iot4pi.com/en/hardware-out-of-the-limelight/lora-raspberry-pi-gateway-en/

Tested with Raspberry 3B+, and Raspberry 4

The config.yaml only needs this insert for the Hat : 

  Module: RF95  # Lora APRS Iot4pi Igate hat with HopeRF 96 (SX1278) 
  Reset: 6
  CS: 8
  IRQ: 5
  Busy: 23 # still testing on the Busy pin to be used, but works as is.

It works with both hopeRF lora chipsets,, 433 and 868 Mhz.

The SPI and I2c  lines in the config.yaml is also made active  , including using sudo raspi-config to enable the SPI and I2C settings.

I2C on this Hat is only used for a BMP-280 sensor, and oled display.

Specify I2C device, or leave blank for none

I2C:
  I2CDevice: /dev/i2c-1
 
Specify the SPI device to use in /dev/. Defaults to spidev0.0
Some devices, like the pinedio, may require spidev0.1 as a workaround.
spidev: spidev0.0

For some reason the service results in an Error after several tries of a rx strange packet,
But this I solved with a workaround and re-starting the service with a crontab every hour.
My node(s) now run for several days without major issues.

Hope this helps others who want to use 'old' Lora APRS Igate boards or make their own (Schematic is included FYI)

[image]

Best regards
Luc Bodson 
ON2ON

My info: { "myNodeNum": 3956045581, "minAppVersion": 30200 }
Metadata: { "firmwareVersion": "2.3.3.8187fa7", "deviceStateVersion": 22, "hasWifi": true, "role": "ROUTER_CLIENT", "positionFlags": 811, "hwModel": "PORTDUINO" }

caveman99, 2024-04-01:
Care to open a PR against the yaml config example file to include your setup?

ON2ON-HAM, 2024-04-01:
I tried to open a PR, but did not succeed ,
I include (if poss) the yaml here for you to use / include

Best Regards
Luc Bodson

closed 2024-04-19 by thebentern