meshtastic/firmware #3533: [Bug]: T-Echo double press

todd-herbert, 2024-04-02:
### Category

Other

### Hardware

T-Echo

### Firmware Version

2.3.4.a4c22321

### Description

Double press on T-echo is assigned to two different actions:
* Toggle backlight
* Send ad-hoc ping

thebentern, 2024-04-03:
Perhaps the capacitive touch button on top would be better suited to the backlight toggle, since the double press ad-hoc ping is available on any platform with a user button

todd-herbert, 2024-04-03:
That's certainly an idea that I can look into. I do remember hearing that there are some design issues with the touch button: that it is triggered every time a LoRa packet is transmitted, but I haven't confirmed this for myself yet.

todd-herbert, 2024-04-05:
@Nikto655 Currently it doesn't do very much.

___

After a bit or experimentation, I think the most user friendly option is to use the touch button as a modifier. 
 
Double press user button to send adhoc ping
Hold touch button + double press user button to toggle backlight.

This both aligns with standard meshtastic behavior, and remains somewhat familiar to T-Echo users.

I should have a pull request for this (and a few other related fixes) shortly.

b8b8, 2024-04-06:
Can we exclude the touch button completely please. the alternate cases i design for the t-echo are not compatible with it and ive pealed it off completely. Needing the touch button over the regular buttons severely limits future designs. Can i ask we add 4x button push for the t-echo light.

closed 2024-04-06 by todd-herbert