meshtastic/firmware #3535: [Bug]: Pi Pico W - unreliable WiFi

madeofstown, 2024-04-02:
### Category

WiFi

### Hardware

Raspberry Pi Pico (W)

### Firmware Version

2.3.3

### Description

Wifi seems to stop working and I am unable to access the device until I reboot the Pico or connect a serial cable. Not sure what could be causing this, it seems to be irregular. Would being in range of an MQTT router just overload the Pico hardware?

GUVWAF, 2024-04-02:
Is it still responding over LoRa when Wi-Fi stops working?

Also, can you post your configuration (e.g. with the CLI `meshtastic --info` after removing personal details)? Especially regarding MQTT: are you using the default MQTT server with default channel LongFast? In that case, indeed it might get overloaded.

madeofstown, 2024-04-02:
Yeah, it appears in the node list of my other devices and I am able to successfully "Request Position" and have it update. I'm just unable to log in over the network.

GUVWAF, 2024-04-02:
Might be related to #3031 then. 
I would advise to try another channel name or a different MQTT root topic and see if the issue persists.

madeofstown, 2024-04-03:
I can confirm that changing from the default root topic has fixed the problem. I guess all those messages were too much for the pico to handle.

GUVWAF, 2024-04-03:
Thanks for confirming. I'm afraid there's not much we can do about it, so I'm closing this.

closed 2024-04-03 by GUVWAF