meshtastic/firmware #3543: [Bug]: T-Deck canned messages only send to primary channel

irreverends, 2024-04-03:
### Category

Hardware Compatibility, Other

### Hardware

T-Deck

### Firmware Version

2.3.2

### Description

Sending a canned message when choosing a node to send it to, seems to only send it to the primary channel.

HarukiToreda, 2024-04-04:
when you press Tab, it allows you to select the node pressing left or right, Press Tab again and it allows you to choose the other channels, just press left and right these show you which you're modifying ><

irreverends, 2024-04-04:
> when you press Tab, it allows you to select the node pressing left or right, Press Tab again and it allows you to choose the other channels, just press left and right these show you which you're modifying ><

Swiping left or right on the screen, or using the track ball changes the node it's supposed to send to at the top, but it still just sends to the primary channel.

thebentern, 2024-04-09:
This will eventually be addressed in the TFT UI rework

closed 2024-04-09 by thebentern

mverch67, 2024-04-13:
This PR #3536 may solve it. It emulates the tab key on T-Deck.