meshtastic/firmware #3546: [Bug]:

weirdal3333, 2024-04-04:
### Category

WiFi

### Hardware

T-Beam

### Firmware Version

2.3.2.63df972

### Description

The esp32 has a hardcoded 40mhz channel width.
This causes MAJOR issues when connecting to a hotspot that only supports 20mhz channels, or has 40mhz channels disabled.
Some users report unstable wifi connections.
My router (ddwrt) has logged over 50,000 dropped packets PER DAY for my tbeam due to this issue. THese devices don't need that much bandwidth, and routers with a 40mhz channel width know how to communicate with simpler devices that don't support it.
[image]
[image]

caveman99, 2024-04-04:
I don't see how we (the meshtastic project) can change a value built into ESP32 silicone (or at least a binary blob wifi firmware). This part of the toolchain is not even open source.

closed 2024-04-04 by caveman99

todd-herbert, 2024-04-04:
Adding that line of code would break a lot of ESP32-S3 devices. If you were to implement 20MHz channel width, you would need to ensure that it is only set for the older ESP32 devices. 

In theory that would be possible, but it makes the assumption that all existing users would be okay with the change. I myself would worry about breaking someone else's particular setup with a seemingly innocent change. If it were to go ahead, I would consider making the 20MHz channel width an opt-in config, as a precaution. 

It may even be enough to wrap the line of code in a macro, allowing advanced users to easily select it when building from source. I imagine that most users aware of the 40MHz / 20MHz distinction are probably capable of building their own firmware, and may already be doing so.