meshtastic/firmware #3549: [Bug]: Messages unencrypted if MQTT enabled on the device

cyberorg, 2024-04-04:
### Category

Other

### Hardware

T-Echo

### Firmware Version

2.3.2

### Description

Steps to reproduce:
1. Set up an encrypted channel
2. Enable mqtt on a device sending messages
3. Check mqtt debug "Text" log on nodered. https://github.com/scruplelesswizard/meshtastic-node-red flow for example.
4. Or however you can read mqtt messages after protobuf decode.

The messages originating from device where mqtt is enabled are visible in plain text/unencrypted on mqtt broker.

GUVWAF, 2024-04-04:
Where is your [`mqtt.encryption_enabled`](https://meshtastic.org/docs/configuration/module/mqtt/#encryption-enabled) setting set to?

cyberorg, 2024-04-04:
It is set to off.

Does setting it to on send unencrypted messages unencrypted and encrypted one’s encrypted? I want to send location, unencrypted channel messages etc. unencrypted to mqtt. Just not the encrypted channel messages.

cyberorg, 2024-04-04:
Where do I specify sending “Position” which we use for live tracking unencrypted?

closed 2024-04-04 by thebentern

GUVWAF, 2024-04-04:
You'd have to remove the key on the primary channel (set it to `none`).

cyberorg, 2024-04-04:
Great, thanks for clarifying. Will send a PR later adding this to the doc.

cyberorg, 2024-04-05:
All messages are encrypted if mqtt encryption is enabled. There doesn’t seem to be any way to keep LongFast unencrypted.

Disabling encryption in mqtt makes even channels with encryption key plain text.

Channel configuration.

[image]

Mqtt configuration 
[image]

Lonfast is encrypted, it should not be.
[image]

GUVWAF, 2024-04-05:
If it doesn't have a key, the "encrypted" part is already a [`Data` message](https://github.com/meshtastic/protobufs/blob/6157a5723745b3a750720b94676198a7f3839e2a/meshtastic/mesh.proto#L697), but that is still a protobuf encoded message. Depending on the portnum, you can decode the payload. You should see the text in it already if you send a text message.

cyberorg, 2024-04-05:
The nodered flow already does decode, when mqtt encryption is not set. This shows up if mqtt encryption is set even on a channel where no key is set.

GUVWAF, 2024-04-05:
Yes, so now you should let the NodeRED flow also decode the part that is "encrypted" if channel is 0, because that doesn't have a key now.

cyberorg, 2024-04-05:
Image attached with "https://maps.app.goo.gl/1XwAtZJavero6p2a8" is after it runs through decode. Let me know if you need access to this nodered and mqtt server if you want to test sending position and text with mqtt encryption enabled and disabled, can share credential.

GUVWAF, 2024-04-05:
You want MQTT encryption enabled (because you want text from other channels end up encrypted on the MQTT server), but then you have to instruct NodeRED to decode the "encrypted" field for channel 0 just like it does when it says "decoded".

cyberorg, 2024-04-05:
Yes, regardless of change on the node to send encrypted or not to mqtt, all messages go through "decode". When encryption is on, it shows Channel 0 without key set messages also encrypted.

Do you mean to say I need to pass those through "decode" twice?

GUVWAF, 2024-04-05:
I’m saying that somewhere the NodeRED flow does not decode the “encrypted” part of the MeshPacket, but it should do that when the channel is 0, because it has no key.

cyberorg, 2024-04-05:
Nodered shouldn't need to decrypt those if they are forwarded to mqtt unencrypted, it would work just as if they do when mqtt encryption is not enabled. "Decode" and "Decrypt" are two different things. All messages go through protobuf decode, no decryption is done in nodered flow.

GUVWAF, 2024-04-05:
I’m not saying it should decrypt them. Yes, NodeRED always decodes the MeshPacket, but apparently it doesn’t decode the “encrypted” part.

If you enable MQTT encryption, the Data message will always show up as “encrypted”, for all channels. However, I told you you have to instruct NodeRED to decode this part of the message as well (using the Data message just like it does when the field is called "decoded") for channel 0.

GUVWAF, 2024-04-05:
Well, that's a different question. Currently it's implemented like this, probably as a safety measure. A packet is never send in the "decoded" state if you enable encryption, it doesn't actually check if it has a key or not.

GUVWAF, 2024-04-05:
Port 24?
After you decoded the "encrypted" field with the `Data` message, it shows which portnumber it is (so e.g. position or text) and then you can decode the `payload`.

cyberorg, 2024-04-05:
alright, let me ask chatgpt.