meshtastic/firmware #3550: [Bug]: User button does not function for T-Beam Supreme @ 2.3.2.63df972

jgspratt, 2024-04-04:
### Category

Hardware Compatibility

### Hardware

T-Beam S3

### Firmware Version

2.3.2.63df972

### Description

The "User" button (middle button) does not seem to do anything.  When I'm connected to the device (so I know the device is powered on) and the screen is off, pressing it does nothing.  When I'm disconnected, it also does nothing.  When the screen is already on, pressing it does nothing.  Short press, long press, double press all do nothing.

Tested on two separate physical devices at the same firmware version paired with different phones.

jgspratt, 2024-04-05:
The issue appears to be with the documentation. The leftmost button (when viewing the screen in a left-to-right / normal orientation) is the User button. I will see if I can update the docs.

closed 2024-04-05 by jgspratt

CyborgySquirrel, 2024-10-23:
have the same problem - the button functionality is compromised. The left more button (with screen on the right) scrolls through the different screens and shuts the device down if held long enough. The middle button does nothing (even options to double click to force send position and triple click to turn off/on GPS) doesn’t work. Button three does resent the device.
Would it please be possible to look into this for us? Not being able to interact with the little screen is frustrating