meshtastic/firmware #3555: [Feature Request]: Add support for LSM6DS3TR-C 6 axis accelerometer / gyroscope

garethhcoleman, 2024-04-05:
### Platform

ESP32

### Description

As used on the unPhone ([details](https://unphone/net) [shop](https://shop.pimoroni.com/products/unphone)) I'd like to introduce this sensor to the Meshtastic firmware.

I've started work and feel reasonably confident it will fit in fairly easily but I'd appreciate any feedback / suggestions you can offer.

There are (at least) two libraries that I'm looking at using, the [adafruit one](https://github.com/adafruit/Adafruit_LSM6DS) and one by [chip makers ST](https://github.com/stm32duino/LSM6DS3). I haven't got shaking working yet using the Adafruit one and considering trying the other one, again advice welcomed.

caveman99, 2024-04-05:
i'd try the adafruit first since many other sensors use the lib stack and this way you avoid code duplication.

jaredquinn, 2024-04-06:
I believe there's a discussion here to have around where this sensor data should go?

I see arguments for and against both as position related data and as telemetry (though not necessarily Environmental Telemetry).

garethhcoleman, 2024-04-06:
I would say our primary or at least initial use case would be to use the
accelerometer for tap to wake or wake on pickup etc.

Telemetry wouldn't be the obvious fit with the unPhone as the device is
more targeting hand held than install up a pole.

Even though it might not be a great fit I wanted to find out if it was
considered best practice to include a telemetry aspect to interface with
that module.

caveman99, 2024-04-07:
raw telemetry is ruled out for any high-rate sensor like these. We only support slow-changing data or data with is averaged over time. Use cases for this type of sensor is heading while stationary (add-on to position module) or the tap-modes already present. Another use-case would be threshold based, so strong deltas would indicate a emergency notification needs to be send.

garethhcoleman, 2024-04-07:
Thanks for the info, seems to make the way forward more straightforward.

garethhcoleman, 2024-04-11:
PR [3593](https://github.com/meshtastic/firmware/pull/3593) submitted

closed 2024-04-12 by thebentern