meshtastic/firmware #3558: [Bug]: heltec missing MQTT subscription?

rFronteddu, 2024-04-05:
### Category

Other

### Hardware

Heltec V3

### Firmware Version

2.3.2.63df972

### Description

I am trying to have an heltec consume messages from MQTT.
* I changed the root to msh/US/rob/snd
* I verified that the device is correctly connected to MQTT by enabling json and uplink and verifying that messages get sent in MQTT
* I disabled uplink and enabled downlink in the primary channel
* from the serial I can see that it subscribes to  
  * msh/US/rob/snd/2/c/LongFast/#
  * msh/US/rob/snd/2/json/LongFast/#

The documentation says that I am supposed to publish into msh/2/json/mqtt/, since I changed the root to msh/US/rob/snd I was expecting to have to publish into msh/US/rob/snd/2/json/mqtt but from the logs it doesn't seem like the device is subscribed to this or similar topics. If I publish in one of the topics the device is subscribed to, I get an error saying that the topic is not the correct or downlink is not enabled. I tried msh/US/rob/snd/2/json/LongFast/mqtt/ amd msh/US/rob/snd/2/json/mqtt/ with no success.

Could you help me figure out what am I missing?

GUVWAF, 2024-04-06:
You have to manually add a channel called "mqtt".

rFronteddu, 2024-04-06:
Thank you. I had to un-power and power back the device for the subscription to happens but it finally works.

closed 2024-04-06 by rFronteddu