meshtastic/firmware #3561: [Feature Request]: long press top button on t-echo for SOS

cyberorg, 2024-04-06:
### Platform

NRF52

### Description

Hi, can we have an SOS assigned to the long(5 seconds) press of top button with a confirmation by touch button?

The SOS would send “SOS - position” on the primary channel.

This would be very helpful for paragliding pilot if they crash and are not in a position to send message themselves.

todd-herbert, 2024-04-06:
One possible issue I see here is that some users may naturally hold the device in a way thay they keep their finger on the touch button during normal use.

Perhaps the SOS feature could be enabled as a module? There could be an option for long press of the touch button, but possibly other keypress options too, like constant clicking of the user button (>10 times?)

cyberorg, 2024-04-06:
That is why 2 steps. Hold side too button 5 seconds(currently unassigned) and when message in screen says, “confirm SOS” tap the top touch button.

thebentern, 2024-04-06:
We are not putting SOS features in the codebase, as this potentially puts us in a place of legal liability

closed 2024-04-06 by thebentern

jp-bennett, 2024-04-06:
There's already a canned messages module that could already be used for this case. Set it up with just the one message, and send it with a couple clicks.

cyberorg, 2024-04-07:
There are many pros for having a default SOS function. Some:

Long press one button and confirmation via any other preconfigured button is easier even with eyes closed. In emergency situations simplest of the methods work best.

It also keeps other functionalities untouched.

SOS is available even if someone forgot to set it up.

SOS once activated can continue broadcasting at 1min interval till an ACK is received.

However if it can’t be done in a default firmware setup then will have a look at Canned messages. However I couldn’t manage to figure it out looking at and trying it. Tried up/down, the buttons continue with their default behaviour, having no impact on selection of message or sending it.

todd-herbert, 2024-04-07:
Having no previous experience with the Canned Message module, I just now attempted to set it up for T-Echo. Within the ~20 minutes I spent, I was not able to achieve a usable config. With some tinkering, it may be possible, but I certainly had difficulty.

The first impression I get is that the module is suited for advanced users, with custom hardware. While any sort of SOS feature is entirely out of the question, I *can* see potential for a Canned Message behavior, which is simpler to set up, and hooks existing buttons, suitable for prebuilt devices such as the T-Echo. 

I could imagine this implemented as either a separate module, or an new input method for the existing module (as mentioned [in the doc](https://meshtastic.org/docs/configuration/module/canned-message/#hardware)). In such a module / input method, I could imagine using either a hardcoded keypress, or allowing some level of customization by users (at their own peril).

One thing that is clear to me, is that any new behavior could not rely solely on the T-Echo's touch button. As I learned in https://github.com/meshtastic/firmware/issues/3533, some users have customized their devices, removing the touch button, as it has historically been unused in the Meshtastic firmware.

cyberorg, 2024-04-07:
> We are not putting SOS features in the codebase, as this potentially puts us in a place of legal liability

Call it “Need help” “help” “assist” or whatever suits legal.

> One thing that is clear to me, is that any new behavior could not rely solely on the T-Echo's touch button.

Use single click of the same button to confirm to keep it simple and would work on any device with buttons.