meshtastic/firmware #3562: [Bug]: json utf-8 characters converted to \n0000

GerritS76, 2024-04-06:
### Category

Other

### Hardware

Heltec V3

### Firmware Version

at least 2.2.4 and above

### Description

When sending a message from my Mestastic node nothing seems to be wrong, however besides this I'm also using a MQTT broker which receives the data as well (in json). When taking a look, characters like éëèê etc. (thus utf-8 chars) are all converted to the same \u0000 in json.

This means it's impossible to (for example) fetch that data 1 on 1 via MQTT and use it for certain purposes.

Also tried this:
`mosquitto_pub -h localhost -u {user} -P {pass} -t 'msh/3300-3329/2/json/mqtt/!{1}' -m '{"channel": 0,"from":{2},"type":"sendtext","payload":"äáàéëèóöò etc. etc."}'`

Which output:

> {"channel":0,"from":{1},"hops_away":0,"id":{2},"payload":{"text":"\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000 etc. etc."},"sender":"!da5c87d4","timestamp":1712428198,"to":4294967295,"type":"text"}

It looks like all data via the Meshtastic node is converted in a bad way causing utf-8 characters beïng replaced with \u0000 (same happens with smilies by the way but that's less important than utf-8 chars).

LOCOSP, 2024-04-26:
@GerritS76 You have ruined my dreams :) opening it again I am waiting for ages for this fix because in my language it is a lot of ąęść 😃

cnstudio, 2024-06-30:
I can confirm this issue also in firmware 2.3.13.83f5ba0 Beta,
look like the MQTT JSON text not decode with UTF-8.

kevinelliott, 2024-07-07:
Can confirm this issue still exists.

My guess is that something in this function is not quite right -- https://github.com/meshtastic/firmware/blob/master/src/mqtt/JSONValue.cpp#L833

kevinelliott, 2024-07-07:
Specifically these lines -- https://github.com/meshtastic/firmware/blob/master/src/mqtt/JSONValue.cpp#L854-L863

kiclaw, 2024-08-09:
I have this problem too, I'm patiently waiting for this to be taken into account.

kevinelliott, 2024-08-10:
I'm away from my computer at the moment, but I'll try to get to this late tonight or early tomorrow. Thanks for waiting.

In the meantime, if you could provide an additional log/stacktrace that would be helpful!

garthvh, 2024-08-13:
The JSON library being used, and the JSON library on device in general is limited and does not support special characters.  You need to use protobufs and parse json externally for full functionality.

closed 2025-02-08 by fifieldt