meshtastic/firmware #3563: [Bug]: Debian package unistallable

kalikid021, 2024-04-06:
### Category

Other

### Hardware

Other

### Firmware Version

2.3.2.63df972

### Description

Raspberry Pi Zero W, with Raspbian OS (Bookworm) 

Deb package has broken dependencies. package was already installed. I removed it, tried to install the deb and allow it to install the dependency, same error, reinstalled dependency, attempted install, same error. It happens using both apt and dpkg

### Relevant log output

```Shell
root@piw-lora:/opt# apt install libyaml-cpp0.7
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following NEW packages will be installed:
  libyaml-cpp0.7
0 upgraded, 1 newly installed, 0 to remove and 0 not upgraded.
Need to get 89.1 kB of archives.
After this operation, 310 kB of additional disk space will be used.
Get:1 http://raspbian.raspberrypi.com/raspbian bookworm/main armhf libyaml-cpp0.7 armhf 0.7.0+dfsg-8+rpi1 [89.1 kB]
Fetched 89.1 kB in 2s (51.4 kB/s)
Selecting previously unselected package libyaml-cpp0.7:armhf.
(Reading database ... 64821 files and directories currently installed.)
Preparing to unpack .../libyaml-cpp0.7_0.7.0+dfsg-8+rpi1_armhf.deb ...
Unpacking libyaml-cpp0.7:armhf (0.7.0+dfsg-8+rpi1) ...
Setting up libyaml-cpp0.7:armhf (0.7.0+dfsg-8+rpi1) ...
Processing triggers for libc-bin (2.36-9+rpt2+deb12u4) ...
root@piw-lora:/opt# apt install libyaml-cpp-dev
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following NEW packages will be installed:
  libyaml-cpp-dev
0 upgraded, 1 newly installed, 0 to remove and 0 not upgraded.
Need to get 137 kB of archives.
After this operation, 845 kB of additional disk space will be used.
Get:1 http://raspbian.raspberrypi.com/raspbian bookworm/main armhf libyaml-cpp-dev armhf 0.7.0+dfsg-8+rpi1 [137 kB]
Fetched 137 kB in 2s (83.4 kB/s)
Selecting previously unselected package libyaml-cpp-dev.
(Reading database ... 64826 files and directories currently installed.)
Preparing to unpack .../libyaml-cpp-dev_0.7.0+dfsg-8+rpi1_armhf.deb ...
Unpacking libyaml-cpp-dev (0.7.0+dfsg-8+rpi1) ...
Setting up libyaml-cpp-dev (0.7.0+dfsg-8+rpi1) ...

root@piw-lora:/opt# apt install ./meshtasticd_2.3.2.63df972_arm64.deb
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Note, selecting 'meshtasticd:arm64' instead of './meshtasticd_2.3.2.63df972_arm64.deb'
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 meshtasticd:arm64 : Depends: libyaml-cpp0.7:arm64 but it is not installable
E: Unable to correct problems, you have held broken packages.
```

jaredquinn, 2024-04-06:
Issue here seems to be architecture - 32bit vs 64bit;  the libyaml-cpp0.7:armhf  vs libyaml-cpp0.7:arm64.

jp-bennett, 2024-04-07:
Yep. You're installing a 64 bit package on a 32 bit Pi install. Making 32 bit binaries is on our radar, but not here yet.

closed 2024-04-07 by kalikid021