meshtastic/firmware #3582: TraceRoute not working thru MQTT on secondary channels

BrunoKlu, 2024-04-09:
### Category

Other

### Hardware

Other

### Firmware Version

2.3.3

### Description

Trying to make traceroutes between 2x  nodes networks linked thru MQTT.

Traceroutes are working on primary channel.

Traces routes are not working on secondary channels.

Traceroutes are working locally but not thru MQTT

I can see the packets going on air with a remote spectrum analyser but the destination node is never answering like if something wrong is going inside the packet. Again if I launch the traceroute localy It is working properly.

GUVWAF, 2024-04-09:
Did you check whether the other side has downlink enabled on this secondary channel as well? Does it receive text messages on this channel, and does it receive direct messages?
Besides that, did you swap or change your channels? Your node needs to know how to reach this node. If it already knew this node before you changed channels, it might not get a new NodeInfo on the right channel. You could try deleting the node, send a message and then it should populate again with the correct channel.

BrunoKlu, 2024-04-09:
Hello, yes both sides are enabled on uplink and downlink on this channel (LongFast). I already tried reseting node DB, I'm doing it again now and test again. Should I do that on every node the traceroute is passing by ? Thanks

BrunoKlu, 2024-04-09:
I recieve messages from this channel and I'm able to send some, will try DM

GUVWAF, 2024-04-09:
> Should I do that on every node the traceroute is passing by ? Thanks

Only the node you're trying to reach should be enough.

BrunoKlu, 2024-04-09:
OK I finaly found what was wrong, from a firmware release, in the LoRa menu the Ignore MQTT seems enabled by default so every node that has just been updated can not communicate back to me ! Takes me 3 days to find it lol .  Thanks for your help

BrunoKlu, 2024-04-09:
Ignore MQTT enabled

closed 2024-04-09 by BrunoKlu