meshtastic/firmware #3586: [Feature Request]: Enhance UX by Auto-Closing or Simplifying Closure of Trace Route Confirmation Popup in OSX

callieschneider, 2024-04-10:
### Platform

ESP32

### Description

This feature request seeks to enhance the user experience surrounding the Trace Route confirmation popup in OSX. Currently, after initiating a Trace Route, a popup appears with the message: "Trace Route Sent - This could take a while, response will appear in the trace route log for the node it was sent to," accompanied by an "OK" button. The proposal aims to streamline this interaction, reducing the need for manual action to close this popup.

Objective:
To simplify and expedite the process for users to continue their work after sending a Trace Route request, by making the closure of the confirmation popup more efficient and less intrusive.

Proposed Enhancements:
1. Automatic Closure: Introduce functionality for the popup to automatically close after a brief, configurable delay. This allows users to be informed that the Trace Route has been initiated without requiring manual action to dismiss the message.
2. Keyboard Shortcut Closure: Enable the popup to be closed by pressing a common keyboard key (e.g., Enter or Esc). This would be particularly useful for users who prefer keyboard navigation and wish to quickly dismiss the popup without reaching for the mouse.
3. Focus on 'OK' Button: Ensure that the "OK" button is automatically focused when the popup appears. This way, users can immediately press Enter to close the popup without having to manually select the button.

Benefits:
- Efficiency: Reduces friction in the user workflow by minimizing unnecessary interactions, allowing users to focus on their tasks.
- User Preference: Offers users multiple methods to dismiss the popup, catering to different preferences and enhancing overall user satisfaction.
- Accessibility: Improves accessibility by supporting keyboard navigation, making the application more inclusive and easier to use for all users.

This enhancement would significantly improve the usability of the Trace Route feature by making the confirmation step smoother and less disruptive.

garthvh, 2024-04-10:
Trace routes are pretty expensive, the confirmation is there to slow down trace routes in rapid succession and increase the chance of success.

callieschneider, 2024-04-10:
I'd considered that as a possibility, makes sense. Thanks for your reply. 👍