meshtastic/firmware #3602: [Bug]: Program button resets node on Heltec V3 with CardKB

HarukiToreda, 2024-04-12:
### Category

Other

### Hardware

Heltec V3

### Firmware Version

2.3.4

### Description

Anytime I press the Program button on the node it resets the node as if i was pressing the reset button. This happens with all standalone Heltec V3 nodes with a CardKB.

### Relevant log output

```Shell
DEBUG | 01:40:30 50 [Router] Add packet record (id=0x0eb72919 fr=0xdc to=0xff, WantAck=0, HopLim=2 Ch=0x4b encrypted rxSNR=5.75 rxRSSI=-22 hopStart=3)

DEBUG | 01:40:30 50 [Router] Ignoring incoming msg, because we've already seen it (id=0x0eb72919 fr=0xdc to=0xff, WantAck=0, HopLim=2 Ch=0x4b encrypted rxSNR=5.75 rxRSSI=-22 hopStart=3)

DEBUG | 01:40:30 50 [Router] cancelSending id=0xeb72919, removed=0

DEBUG | 01:40:30 50 [Router] Incoming message was filtered 0xe2e590dc

abort() was called at PC 0x4037a41f on core 1

Backtrace: 0x4037845e:0x3fc9b590 0x40380ce5:0x3fc9b5b0 0x40387881:0x3fc9b5d0 0x4037a41f:0x3fc9b650 0x4037a531:0x3fc9b680 0x4037a600:0x3fc9b6a0 0x42133681:0x3fc9b6d0 0x42124335:0x3fc9b6f0 0x421243b1:0x3fc9b720 0x4212d8db:0x3fc9b750 0x4212d8ad:0x3fc9b7b0 0x42124d60:0x3fc9b7d0 0x4200b88b:0x3fc9b800 0x4200b8c8:0x3fc9b830 0x4200b8fe:0x3fc9b850 0x42005773:0x3fc9b870 0x42010852:0x3fc9b8e0 0x42010880:0x3fc9b910 0x421428a9:0x3fc9b930 0x403759b1:0x3fc9b950 0x403759d6:0x3fc9b970 0x4037a18d:0x3fc9b990 0x4214297f:0x3fcf5380 0x420732c1:0x3fcf53a0 0x403821dc:0x3fcf53c0

ELF file SHA256: c96d9d6e075bb841

E (8411) esp_core_dump_flash: Core dump flash config is corrupted! CRC=0x7bd5c66f instead of 0x0

Rebooting...

ESP-ROM:esp32s3-20210327

Build:Mar 27 2021

rst:0xc (RTC_SW_CPU_RST),boot:0x28 (SPI_FAST_FLASH_BOOT)

Saved PC:0x42142982

SPIWP:0xee

mode:DIO, clock div:1

load:0x3fce3808,len:0x44c

load:0x403c9700,len:0xbe4

load:0x403cc700,len:0x2a38

entry 0x403c98d4

E (356) esp_core_dump_flash: No core dump partition found!

E (356) esp_core_dump_flash: No core dump partition found!��@INFO | ??:??:?? 0

//\ E S H T /\ S T / C

INFO | ??:??:?? 0 Booted, wake cause 0 (boot count 1), reset_reason=reset

DEBUG | ??:??:?? 0 Filesystem files (520192/1048576 Bytes):

DEBUG | ??:??:?? 0 /prefs/channels.proto (140 Bytes)

DEBUG | ??:??:?? 0 /prefs/config.proto (146 Bytes)

DEBUG | ??:??:?? 0 /prefs/db.proto (935 Bytes)
```

todd-herbert, 2024-04-12:
I saw the same error a few days ago. I believe it's just a result of changes in meshtastic/protobufs.
Try erasing the flash then re-uploading. That fixed it for me.

HarukiToreda, 2024-04-12:
> I saw the same error a few days ago. I believe it's just a result of changes in meshtastic/protobufs. Try erasing the flash then re-uploading. That fixed it for me.

I did a factory reset and set all settings back manually. I was able to pinpoint which setting makes that issue. It is the Up/Down/Select input enabled on Canned Message Config. Once you enable it it causes that issue, if I dissable it it goes back to normal working.

todd-herbert, 2024-04-13:
When you enabled canned messages, did you set the GPIO pins? If they were left at the default of GPIO 0, I could imagine this causing the crash you experienced.

HarukiToreda, 2024-04-13:
> When you enabled canned messages, did you set the GPIO pins? If they were left at the default of GPIO 0, I could imagine this causing the crash you experienced.

I've never had to set them in the past with a cardKB so I didn't think of changing it from GPIO 0, it uses pins 41 and 42 as i2c. 
I just tried setting it from 0 to 41 and 42 and the cardKB stops working completely and the program button still makes it reboot. I think the pins are only for the rotary encoder to work.

HarukiToreda, 2024-04-13:
I'll just avoid the up and down setting for now

todd-herbert, 2024-04-13:
I don't really know anything about the Canned Message module. I get that you've maybe moved on now, but I'm interested to know if you set the input source to cardkb?

https://meshtastic.org/docs/configuration/module/canned-message/#input-source

HarukiToreda, 2024-04-13:
yeah I have the input set for cardKB. I did notice that dispite the setting for "Up/Down/Select input enabled" is dissabled the functionality that it's supposed to do which is to scroll through preset canned messages is still working. I may not need to have that setting turned on after all. I do wonder why it's even there if it works without it. turned on.

todd-herbert, 2024-04-13:
It does sound a bit odd. I would look into it for you, but I don't have the right hardware. Hopefully someone else is able to investigate.

PR #3612 (2024-04-14): fix: UpDown CARDKB button press
fixes #3602

Note: this fix requires another OSThread, watch out if MAX_THREADS (default=32) is large enough for all devices.
closed 2024-04-14 by thebentern