meshtastic/firmware #3608: [Feature Request]: raw packet output mode/option

ianmcorvidae, 2024-04-12:
### Platform

NRF52, ESP32

### Description

I'm not sure the best way to enable this, but what I'd like to be able to do is get a stream of _all_ packets the radio sees in some fashion (not just those on a known channel/with a known encryption key). The use case I have in mind is multi-radio setups consisting of a couple nodes on different presets and/or frequency slots, all connected to a separate MCU, mini-pc/SBC, or similar. Such a setup could in theory be used, for example, to shuttle packets over a faster preset or less congested frequency slot among known, fixed nodes, while the main/overall mesh is still using default settings, to support faster propagation of packets and (hopefully) reduced airtime overall on the default settings.

I would ideally like to be able to make something that works more like a single radio in that it's capable of rebroadcasting packets it doesn't have the key for, but at present the only ways to get packets out of a radio are the phone APIs (which don't send anything non-decryptable) and MQTT (which doesn't get sent anything non-decryptable), so at present such a setup would be limited to known channels/keys only.

garthvh, 2024-04-16:
You can use the protobuf output from the serial module, or build a client using the protobufs api

closed 2024-04-16 by garthvh

ianmcorvidae, 2024-04-16:
According to @thebentern in Discord (and as I thought was the case), the packets I'm asking for a way to access are not included in either serial output or the protobuf API: https://discord.com/channels/867578229534359593/919642584480112750/1228151081943564288

While the serial module and protobuf API are probably usable for this on known channels, I want to see packets the firmware can't decrypt as well.

It does also occur to me that the other end of this for a bridge-like setup is _also_ probably not currently possible, i.e., directly passing an encrypted packet for which the firmware does not know the key and having it be broadcast.