meshtastic/firmware #3618: [Bug]: Can't communicate via Ethernet

micheljung, 2024-04-14:
### Category

Other

### Hardware

Rak4631

### Firmware Version

2.3.4.ea61808

### Description

I wrote a Software that successfully connects via Ethernet and retrieves device information. However, when trying to send a chat message, the device drops the connection.

To make sure the problem is not on my end, I tried the Python CLI and it seems to confirm the issue:

```
$ meshtastic --version
2.3.4

$ meshtastic --host 192.168.178.47 --info
Error connecting to 192.168.178.47:Timed out waiting for connection completion
```

### Relevant log output

```Shell
INFO | 10:55:09 23 [ApiServer] Incoming wifi connection
INFO | 10:55:09 23 [ApiServer] Incoming ethernet connection
INFO | 10:55:09 23 [ServerAPI] Client wants config, nonce=1024315206
INFO | 10:55:09 23 [ServerAPI] Starting API client config
INFO | 10:55:09 23 [ServerAPI] getFromRadio=STATE_SEND_MY_INFO
DEBUG | 10:55:09 23 [ServerAPI] encoding toPhone packet to phone variant=3, 12 bytes
INFO | 10:55:09 23 [ServerAPI] getFromRadio=STATE_SEND_METADATA
DEBUG | 10:55:09 23 [ServerAPI] encoding toPhone packet to phone variant=13, 30 bytes
INFO | 10:55:09 23 [ServerAPI] getFromRadio=STATE_SEND_NODEINFO
INFO | 10:55:09 23 [ServerAPI] nodeinfo: num=0xcdaf65a1, lastseen=1713092109, id=!cdaf65a1, name=KP Hotzenweid
DEBUG | 10:55:09 23 [ServerAPI] encoding toPhone packet to phone variant=4, 95 bytes
INFO | 10:55:09 23 [ServerAPI] getFromRadio=STATE_SEND_NODEINFO
INFO | 10:55:09 23 [ServerAPI] nodeinfo: num=0xe5e837d0, lastseen=0, id=!e5e837d0, name=T-Echo #1
DEBUG | 10:55:09 23 [ServerAPI] encoding toPhone packet to phone variant=4, 46 bytes
INFO | 10:55:09 23 [ServerAPI] getFromRadio=STATE_SEND_NODEINFO
INFO | 10:55:09 23 [ServerAPI] Done sending nodeinfos
INFO | 10:55:09 23 [ServerAPI] getFromRadio=STATE_SEND_CHANNELS
DEBUG | 10:55:09 23 [ServerAPI] encoding toPhone packet to phone variant=10, 13 bytes
INFO | 10:55:09 23 [ServerAPI] getFromRadio=STATE_SEND_CHANNELS
DEBUG | 10:55:09 23 [ServerAPI] encoding toPhone packet to phone variant=10, 6 bytes
INFO | 10:55:09 23 [ServerAPI] getFromRadio=STATE_SEND_CHANNELS
DEBUG | 10:55:09 23 [ServerAPI] encoding toPhone packet to phone variant=10, 6 bytes
INFO | 10:55:09 23 [ServerAPI] getFromRadio=STATE_SEND_CHANNELS
DEBUG | 10:55:09 23 [ServerAPI] encoding toPhone packet to phone variant=10, 6 bytes
INFO | 10:55:09 23 [ServerAPI] getFromRadio=STATE_SEND_CHANNELS
DEBUG | 10:55:09 23 [ServerAPI] encoding toPhone packet to phone variant=10, 6 bytes
INFO | 10:55:09 23 [ServerAPI] getFromRadio=STATE_SEND_CHANNELS
DEBUG | 10:55:09 23 [ServerAPI] encoding toPhone packet to phone variant=10, 6 bytes
INFO | 10:55:09 23 [ServerAPI] getFromRadio=STATE_SEND_CHANNELS
DEBUG | 10:55:09 23 [ServerAPI] encoding toPhone packet to phone variant=10, 6 bytes
INFO | 10:55:09 23 [ServerAPI] getFromRadio=STATE_SEND_CHANNELS
DEBUG | 10:55:09 23 [ServerAPI] encoding toPhone packet to phone variant=10, 6 bytes
INFO | 10:55:10 23 [ServerAPI] getFromRadio=STATE_SEND_CONFIG
DEBUG | 10:55:10 23 [ServerAPI] encoding toPhone packet to phone variant=5, 9 bytes
INFO | 10:55:10 23 [ServerAPI] getFromRadio=STATE_SEND_CONFIG
DEBUG | 10:55:10 23 [ServerAPI] encoding toPhone packet to phone variant=5, 20 bytes
INFO | 10:55:10 23 [ServerAPI] getFromRadio=STATE_SEND_CONFIG
DEBUG | 10:55:10 23 [ServerAPI] encoding toPhone packet to phone variant=5, 17 bytes
INFO | 10:55:10 23 [ServerAPI] getFromRadio=STATE_SEND_CONFIG
DEBUG | 10:55:10 23 [ServerAPI] encoding toPhone packet to phone variant=5, 22 bytes
INFO | 10:55:10 24 [ServerAPI] getFromRadio=STATE_SEND_CONFIG
DEBUG | 10:55:10 24 [ServerAPI] encoding toPhone packet to phone variant=5, 7 bytes
INFO | 10:55:10 24 [ServerAPI] getFromRadio=STATE_SEND_CONFIG
DEBUG | 10:55:10 24 [ServerAPI] encoding toPhone packet to phone variant=5, 19 bytes
INFO | 10:55:10 24 [ServerAPI] getFromRadio=STATE_SEND_CONFIG
DEBUG | 10:55:10 24 [ServerAPI] encoding toPhone packet to phone variant=5, 12 bytes
INFO | 10:55:10 24 [ServerAPI] getFromRadio=STATE_SEND_MODULECONFIG
DEBUG | 10:55:10 24 [ServerAPI] encoding toPhone packet to phone variant=9, 60 bytes
INFO | 10:55:10 24 [ServerAPI] getFromRadio=STATE_SEND_MODULECONFIG
DEBUG | 10:55:10 24 [ServerAPI] encoding toPhone packet to phone variant=9, 4 bytes
INFO | 10:55:10 24 [ServerAPI] getFromRadio=STATE_SEND_MODULECONFIG
DEBUG | 10:55:10 24 [ServerAPI] encoding toPhone packet to phone variant=9, 17 bytes
INFO | 10:55:10 24 [ServerAPI] getFromRadio=STATE_SEND_MODULECONFIG
DEBUG | 10:55:10 24 [ServerAPI] encoding toPhone packet to phone variant=9, 4 bytes
INFO | 10:55:10 24 [ServerAPI] getFromRadio=STATE_SEND_MODULECONFIG
DEBUG | 10:55:10 24 [ServerAPI] encoding toPhone packet to phone variant=9, 4 bytes
INFO | 10:55:10 24 [ServerAPI] getFromRadio=STATE_SEND_MODULECONFIG
DEBUG | 10:55:10 24 [ServerAPI] encoding toPhone packet to phone variant=9, 13 bytes
INFO | 10:55:10 24 [ServerAPI] getFromRadio=STATE_SEND_MODULECONFIG
DEBUG | 10:55:10 24 [ServerAPI] encoding toPhone packet to phone variant=9, 4 bytes
INFO | 10:55:10 24 [ServerAPI] getFromRadio=STATE_SEND_MODULECONFIG
DEBUG | 10:55:10 24 [ServerAPI] encoding toPhone packet to phone variant=9, 4 bytes
INFO | 10:55:10 24 [ServerAPI] getFromRadio=STATE_SEND_MODULECONFIG
DEBUG | 10:55:10 24 [ServerAPI] encoding toPhone packet to phone variant=9, 4 bytes
INFO | 10:55:10 24 [ServerAPI] getFromRadio=STATE_SEND_MODULECONFIG
DEBUG | 10:55:10 24 [ServerAPI] encoding toPhone packet to phone variant=9, 7 bytes
INFO | 10:55:10 24 [ServerAPI] getFromRadio=STATE_SEND_MODULECONFIG
DEBUG | 10:55:10 24 [ServerAPI] encoding toPhone packet to phone variant=9, 14 bytes
INFO | 10:55:10 24 [ServerAPI] getFromRadio=STATE_SEND_MODULECONFIG
DEBUG | 10:55:10 24 [ServerAPI] encoding toPhone packet to phone variant=9, 8 bytes
INFO | 10:55:10 24 [ServerAPI] getFromRadio=STATE_SEND_MODULECONFIG
DEBUG | 10:55:10 24 [ServerAPI] encoding toPhone packet to phone variant=9, 4 bytes
INFO | 10:55:11 24 [ServerAPI] getFromRadio=STATE_SEND_COMPLETE_ID
DEBUG | 10:55:11 24 [ServerAPI] encoding toPhone packet to phone variant=7, 6 bytes
INFO | 10:55:11 24 [ServerAPI] getFromRadio=STATE_SEND_PACKETS
DEBUG | 10:55:11 24 [ServerAPI] encoding toPhone packet to phone variant=11, 6 bytes
DEBUG | 10:55:11 24 [GPS] Probing for GPS at 115200
WARN | 10:55:12 26 [GPS] Failed to find UBlox & MTK GNSS Module using baudrate 115200
INFO | 10:55:12 26 [ApiServer] Force closing previous TCP connection
INFO | 10:55:13 27 [ApiServer] Incoming wifi connection
INFO | 10:55:13 27 [ApiServer] Incoming ethernet connection
DEBUG | 10:55:13 27 [Power] Battery: usbPower=1, isCharging=1, batMv=4011, batPct=83
INFO | 10:55:13 27 [ServerAPI] Client dropped connection, suspending API service
```

micheljung, 2024-04-14:
This issue does not occur when starting the firmware in a Docker container:

```
$ meshtastic --host localhost
Connected to radio
```

Log:
```
INFO  | ??:??:?? 21 [ApiServer] Incoming wifi connection
INFO  | ??:??:?? 21 [ServerAPI] Client wants config, nonce=2920483757
INFO  | ??:??:?? 21 [ServerAPI] Starting API client config
INFO  | ??:??:?? 21 [ServerAPI] getFromRadio=STATE_SEND_MY_INFO
DEBUG | ??:??:?? 21 [ServerAPI] encoding toPhone packet to phone variant=3, 12 bytes
INFO  | ??:??:?? 21 [ServerAPI] getFromRadio=STATE_SEND_METADATA
DEBUG | ??:??:?? 21 [ServerAPI] encoding toPhone packet to phone variant=13, 18 bytes
INFO  | ??:??:?? 21 [ServerAPI] getFromRadio=STATE_SEND_NODEINFO
INFO  | ??:??:?? 21 [ServerAPI] nodeinfo: num=0x20981da7, lastseen=0, id=!20981da7, name=Meshtastic 0001
DEBUG | ??:??:?? 21 [ServerAPI] encoding toPhone packet to phone variant=4, 58 bytes
INFO  | ??:??:?? 21 [ServerAPI] getFromRadio=STATE_SEND_NODEINFO
INFO  | ??:??:?? 21 [ServerAPI] Done sending nodeinfos
INFO  | ??:??:?? 21 [ServerAPI] getFromRadio=STATE_SEND_CHANNELS
DEBUG | ??:??:?? 21 [ServerAPI] encoding toPhone packet to phone variant=10, 13 bytes
INFO  | ??:??:?? 21 [ServerAPI] getFromRadio=STATE_SEND_CHANNELS
DEBUG | ??:??:?? 21 [ServerAPI] encoding toPhone packet to phone variant=10, 6 bytes
INFO  | ??:??:?? 21 [ServerAPI] getFromRadio=STATE_SEND_CHANNELS
DEBUG | ??:??:?? 21 [ServerAPI] encoding toPhone packet to phone variant=10, 6 bytes
INFO  | ??:??:?? 21 [ServerAPI] getFromRadio=STATE_SEND_CHANNELS
DEBUG | ??:??:?? 21 [ServerAPI] encoding toPhone packet to phone variant=10, 6 bytes
INFO  | ??:??:?? 21 [ServerAPI] getFromRadio=STATE_SEND_CHANNELS
DEBUG | ??:??:?? 21 [ServerAPI] encoding toPhone packet to phone variant=10, 6 bytes
INFO  | ??:??:?? 21 [ServerAPI] getFromRadio=STATE_SEND_CHANNELS
DEBUG | ??:??:?? 21 [ServerAPI] encoding toPhone packet to phone variant=10, 6 bytes
INFO  | ??:??:?? 21 [ServerAPI] getFromRadio=STATE_SEND_CHANNELS
DEBUG | ??:??:?? 21 [ServerAPI] encoding toPhone packet to phone variant=10, 6 bytes
INFO  | ??:??:?? 21 [ServerAPI] getFromRadio=STATE_SEND_CHANNELS
DEBUG | ??:??:?? 21 [ServerAPI] encoding toPhone packet to phone variant=10, 6 bytes
INFO  | ??:??:?? 21 [ServerAPI] getFromRadio=STATE_SEND_CONFIG
DEBUG | ??:??:?? 21 [ServerAPI] encoding toPhone packet to phone variant=5, 11 bytes
INFO  | ??:??:?? 21 [ServerAPI] getFromRadio=STATE_SEND_CONFIG
DEBUG | ??:??:?? 21 [ServerAPI] encoding toPhone packet to phone variant=5, 20 bytes
INFO  | ??:??:?? 21 [ServerAPI] getFromRadio=STATE_SEND_CONFIG
DEBUG | ??:??:?? 21 [ServerAPI] encoding toPhone packet to phone variant=5, 17 bytes
INFO  | ??:??:?? 21 [ServerAPI] getFromRadio=STATE_SEND_CONFIG
DEBUG | ??:??:?? 21 [ServerAPI] encoding toPhone packet to phone variant=5, 20 bytes
INFO  | ??:??:?? 21 [ServerAPI] getFromRadio=STATE_SEND_CONFIG
DEBUG | ??:??:?? 21 [ServerAPI] encoding toPhone packet to phone variant=5, 7 bytes
INFO  | ??:??:?? 21 [ServerAPI] getFromRadio=STATE_SEND_CONFIG
DEBUG | ??:??:?? 21 [ServerAPI] encoding toPhone packet to phone variant=5, 14 bytes
INFO  | ??:??:?? 21 [ServerAPI] getFromRadio=STATE_SEND_CONFIG
DEBUG | ??:??:?? 21 [ServerAPI] encoding toPhone packet to phone variant=5, 12 bytes
INFO  | ??:??:?? 21 [ServerAPI] getFromRadio=STATE_SEND_MODULECONFIG
DEBUG | ??:??:?? 21 [ServerAPI] encoding toPhone packet to phone variant=9, 53 bytes
INFO  | ??:??:?? 21 [ServerAPI] getFromRadio=STATE_SEND_MODULECONFIG
DEBUG | ??:??:?? 21 [ServerAPI] encoding toPhone packet to phone variant=9, 4 bytes
INFO  | ??:??:?? 21 [ServerAPI] getFromRadio=STATE_SEND_MODULECONFIG
DEBUG | ??:??:?? 21 [ServerAPI] encoding toPhone packet to phone variant=9, 4 bytes
INFO  | ??:??:?? 21 [ServerAPI] getFromRadio=STATE_SEND_MODULECONFIG
DEBUG | ??:??:?? 21 [ServerAPI] encoding toPhone packet to phone variant=9, 4 bytes
INFO  | ??:??:?? 21 [ServerAPI] getFromRadio=STATE_SEND_MODULECONFIG
DEBUG | ??:??:?? 21 [ServerAPI] encoding toPhone packet to phone variant=9, 4 bytes
INFO  | ??:??:?? 21 [ServerAPI] getFromRadio=STATE_SEND_MODULECONFIG
DEBUG | ??:??:?? 21 [ServerAPI] encoding toPhone packet to phone variant=9, 13 bytes
INFO  | ??:??:?? 21 [ServerAPI] getFromRadio=STATE_SEND_MODULECONFIG
DEBUG | ??:??:?? 21 [ServerAPI] encoding toPhone packet to phone variant=9, 4 bytes
INFO  | ??:??:?? 21 [ServerAPI] getFromRadio=STATE_SEND_MODULECONFIG
DEBUG | ??:??:?? 21 [ServerAPI] encoding toPhone packet to phone variant=9, 4 bytes
INFO  | ??:??:?? 21 [ServerAPI] getFromRadio=STATE_SEND_MODULECONFIG
DEBUG | ??:??:?? 21 [ServerAPI] encoding toPhone packet to phone variant=9, 4 bytes
INFO  | ??:??:?? 21 [ServerAPI] getFromRadio=STATE_SEND_MODULECONFIG
[... 350 lines cut ...]
DEBUG | 12:29:52 23 [SimRadio] AirTime - Packet transmitted : 419ms
DEBUG | 12:29:52 23 [SimRadio] Completed sending (id=0xddf801d9 fr=0xa7 to=0xff, WantAck=0, HopLim=3 Ch=0x8 encrypted hopStart=3 priority=64)
INFO  | 12:29:58 30 [NodeInfoModule] Sending our nodeinfo to mesh (wantReplies=1)
INFO  | 12:29:58 30 [NodeInfoModule] sending owner !20981da7/Meshtastic 0001/0001
DEBUG | 12:29:58 30 [NodeInfoModule] Update DB node 0x20981da7, rx_time=1713097798
DEBUG | 12:29:58 30 [NodeInfoModule] handleReceived(LOCAL) (id=0x36ad4ec7 fr=0xa7 to=0xff, WantAck=0, HopLim=3 Ch=0x0 Portnum=4 WANTRESP rxtime=1713097798 priority=10)
DEBUG | 12:29:58 30 [NodeInfoModule] No modules interested in portnum=4, src=LOCAL
DEBUG | 12:29:58 30 [NodeInfoModule] localSend to channel 0
[... cut ...]
DEBUG | 12:29:59 30 [SimRadio] decoded message (id=0x36ad4ec7 fr=0xa7 to=0xff, WantAck=0, HopLim=3 Ch=0x0 Portnum=4 WANTRESP rxtime=1713097798 hopStart=3 priority=10)
DEBUG | 12:29:59 30 [SimRadio] (bw=250, sf=11, cr=4/5) packet symLen=8 ms, payloadSize=66, time 722 ms
DEBUG | 12:29:59 30 [SimRadio] AirTime - Packet transmitted : 722ms
DEBUG | 12:29:59 31 [SimRadio] Completed sending (id=0x36ad4ec7 fr=0xa7 to=0xff, WantAck=0, HopLim=3 Ch=0x8 encrypted rxtime=1713097798 hopStart=3 priority=10)
```

daytonturner, 2024-05-01:
OK, Ive spent a couple hours poking at this and I think I've got some answers on why this is happening, but I don't know the best approach for solving it.

Taking a look at how the TCP API loop is implemented here: https://github.com/meshtastic/firmware/blob/master/src/mesh/api/ServerAPI.cpp#L48-L59

My understanding is this:
- Line 48: calling the RAK EthernetServer::available (client), will return an instance of EthernetClient if an existing connection is established and has contents in the recv buffer ... AKA, a new connection has come in.
- Line 49: OK, there is a new connection, lets begin.
- Line 51/53: If there's currently an open TCP API connection, destroy it
- Line 56: If there isn't an actiive TCP API connection, create it
- Line 59: Wait 100ms and restart the loop

The issue of course, is a client connected to the TCP API will send a number of requests through the life of the connection, so the receive buffer will have data in it much longer than the 100ms it takes to start the loop over.

When the loop repeats in the next 100ms, Line 48/49 will return true, because there is an established connection, likely with contents in the recv buffer.  The connection we made a moment ago.  Its about to be killed.

This generally seems like an incomplete way to handle this.  I went looking through old git commits and found #608 where the force-closing of the previous TCP connection was introduced.  I get that we only want to permit one connection at a time, but outright killing the existing connection isn't a valid route either.

This behavior makes some sense if you only consider long-running connections that have a 0-length receive buffer waiting from the client (such as the Android app idling, some other custom application watching for received messages, etc).

However, if you're using the CLI via the TCP API,  queries can take ~3 seconds to fully complete, and given the 100ms loop, there's no chance they'll finish before they get auto-destroyed the next time the loop runs.  I've temporarily changed `return 100;` to `return 4000;` to allow enough time for the CLI to do its thing, but thats not a proper solution either.

Hoping someone has some other ideas on ways to handle the various edge cases here?

daytonturner, 2024-05-01:
Thinking through some high level logic on an appropriate way to handle this, and I have an idea to propose.

When a new connection comes in, the code currently checks to see if there are any existing clients (openAPI) and if so, it destroys them.  I feel like the intention here was to only behave this way if the connection is a long-running, currently idle established connection.  

Perhaps on line 51, rather than blindly checking for an existing client, we should be evaluating the currently established client a bit further, and checking if its receive buffer is empty or not.  If it is, allow it to be killed.  This would allow a second long-running client to "take over" the single connection that's allwoed.  If there is data in the recv buffer, dont destroy the client, but also dont attempt to create a new one.  This would allow an existing client that's still actively asking for data to finish.

I can see how this could be accomplished with the private functions within EthernetServer/EthernetClient, but not entirely sure if the public functions allow for this level of inspection.  I'm not a C++ developer, so I'm just taking a shot here.

Another thought of course, is if you have two clients (let's use the android app using IP connections as our example) both trying to connect to the same API, and the server doesn't want to allow more than one.. If you let the server kill an existing connection and give the new client the socket, it seems like this will actually just cause the two clients to fight while they repeatedly attempt to reconnect - ultimately making neither of them function.  I dont know how many times they retry, but this doesn't seem viable either.  Perhaps closing the existing connection shouldn't happen at all, and we should just reject new connections with an error if one already exists?

daytonturner, 2024-05-03:
Interestingly - I've updated to the latest python meshtastic v2.3.8 and noticed that there's a major change in it to use the 'modern' meshtastic protocol init flow.  In the PR it mentions timeouts when using the TCP API:

https://github.com/meshtastic/python/pull/560

After this update, the CLI seems to work basically 100% of the time.  It makes some sense, given that it can accomplish the task much faster, but its also entirely confusing given how i know the code mentioned above should work, so there must still be something about the TCP API loop that I dont understand. 

Will try to get a better understanding of how that loop works, but for now if you update the meshtastic CLI to 2.3.8, i suspect you'll also see that its now working reliably.  Curious to hear your results.

daytonturner, 2024-05-05:
> @daytonturner<https://github.com/daytonturner> they way I understand it, meshtastic/python#560<https://github.com/meshtastic/python/pull/560> only eliminates the symptom, not the issue.
>
> Before, the Python CLI did:
>
>   1.  Establish a connection (works)
>   2.  Request channels (failed, because of this bug)
>
> Now, it only does 1., which works, and 2. is no longer needed.
>
> In the client I'm writing, I can also successfully establish a connection and receive the device information. However, as soon as I send another message to the device, the connection is dropped.

It makes sense that this behaviour would mask the problem, rather than fixing the root of the problem.

I say this because the main ServerAPI loop checks to see if a new connection is waiting to be served by testing if there’s an established connection with bytes waiting in the receive buffer (ie the client hasn’t sent its request yet).

If the client makes only one request and is simply waiting for response to happen, then it won’t be a problem.

If after 100ms (the loop time) there’s still a client connected with anything in the receive buffer (like a second request) it will close it and open a new one.

Dayton

NetHorizon82, 2024-09-05:
I think that I am having the same issue, but I am not familiar enough to get this far into the weeds yet. I was able to get it to send messages to a private MQTT server so I know that the device is connected to the network. However, if I try to use the web client to connect the devices IP, the client just never fully loads.

garthvh, 2024-09-06:
There is no web server on a nrf device you need to use the hosted web client at client.meshtastic.org

NetHorizon82, 2024-09-06:
> There is no web server on a nrf device you need to use the hosted web client at client.meshtastic.org

I know that the RAK4631 does not self host the client. I used both the client.meshtastic.org as well as a client I host locally on another device.

In both cases, when I connect to the device using the IP:
     1. All the device info in the upper left corner (short name, long name, battery, firmware ver.) shows up as "UNK"
     2. The messages window says "Messages: Loading..." and never changes.
     3. clicking on config get the basic config screen with all fields blank.

Everything works fine over serial, and I have a RPi/Waveshare node that I access over the same network so I doubt that there is a problem with the web host/network itself.

closed 2024-09-06 by thebentern

EmilAndersson-SE, 2024-09-06:
How can this be completed? Is the issue fixed and in what firmware?

logikal, 2024-09-06:
It looks like https://github.com/meshtastic/firmware/pull/4639 addressed this.

caveman99, 2024-09-06:
as soon as we merge the acoomanying pull request the issue will automatically be closed. that means the fix is included in the next alpha release.

NetHorizon82, 2024-09-10:
I tried the new firmware, but am still having the same results when using the web client to connect. Android client (connecting over ethernet) does seem to work now. Is there someone I can work with to pull logs to see if this is still the same issue or if there is something else going on?

zerolint, 2024-09-15:
@NetHorizon82 The web client uses a nodes http/s web server, not the port 4403 api server.
Currently the web client will not work connecting to a rak4631 with Ethernet since only a port 4403 api server is run.

NetHorizon82, 2024-09-18:
@zerolint Thank you for the info. Is only running the API server and not both API and http/s a hardware limitation or just something that hasn't been implemented due to priorities? Also, if it is a hardware RESOURCE limitation, would it be possible to make it so that someone could choose if they wanted the API or http/s server to be running?

Also, I am new to participating, so please let me know if there is a better place to discus this issue other than a close bug report.

arne182, 2024-10-18:
Just updated to 2.5.5 and found that the ethernet is no longer working. Is there any update about what version still works?