meshtastic/firmware #3619: [Bug]: Dockerfile: --hwid: (PROGRAM ERROR) Option should have been recognized!?

micheljung, 2024-04-14:
### Category

Other

### Hardware

Not Applicable

### Firmware Version

n/a

### Description

If you follow [Usage with Docker](https://meshtastic.org/docs/software/linux-native/#usage-with-docker) and execute:

    docker run --rm -p 4403:4403 meshtastic/device

You get:

    meshtasticd: --hwid: (PROGRAM ERROR) Option should have been recognized!?

Because of this in the Dockerfile:

    --hwid=${HWID:-$RANDOM}

But neither of these variables are set and I'm not sure how this is supposed to work. I assume `HWID` should be a build argument but what about `RANDOM`?

ymilord, 2024-04-17:
Ran into the same issue. How do I go about fixing this?

micheljung, 2024-04-17:
@ymilord remove `--hwid=${HWID:-$RANDOM}` from the Dockerfile and build the image again

closed 2024-11-07 by github-actions[bot]