meshtastic/firmware #3620: [Feature Request]: Node Broadcast Info - See FW Version

SubnetConflict, 2024-04-14:
### Platform

ESP32

### Description

Node Broadcast info - Would it be possible (Not with MQTT) to get the remote nodes firmware version visible, on the nodes list, please?

Perhaps push notifications on the app as well for Minor releases and Major releases for the node owners? 
I am curious and I'm sure others would be curious to see how many nodes in their areas are running ancient firmware.
Perhaps make the major releases somewhat enforced? 
With how rapidly the development team works it would be good to try and keep everybody at least on the latest 'Major' update. 
I'm sure there's loads of behind-the-scenes changes have been made in the last few releases which directly help the mesh overall, so if we have people with old Firmware they could be ruining the experience for the rest.

Thanks

SubnetConflict

garthvh, 2024-04-15:
Firmware version is available on the device metadata admin packet, it is not broadcast.

closed 2024-04-15 by garthvh

thebentern, 2024-04-15:
I do want to add firmware version to the regular broadcasts in 3.0, but as it stands, the string of the full firmware version is a really poor wire format for that data.

SubnetConflict, 2024-04-15:
> I do want to add firmware version to the regular broadcasts in 3.0, but as it stands, the string of the full firmware version is a really poor wire format for that data.

I'm sure it could be trimmed down to the first 3 sets of digits or first 2 sets of digits, it would be useful I think, so if we see an out of date node we can contact the user and ask them to upgrade.

slash-bit, 2024-04-15:
I think 2 first digits of the  version would be sufficient. So for example 2.3.2 would be sent as string 23
Shouldn't take that much space...

asjmcguire, 2024-04-15:
Could I suggest, rather than shoehorning the firmware version in to the telemetry packet, what if we looked at a new packet type, that was broadcast maybe once every 12 hours. Something like a network health or stats packet. Instead of uptime for example being shoved into telemetry - this packet could broadcast uptime, firmware version and maybe stats on how busy the node has been, something like a count of the different packet types. This would enable spotting what nodes are finding the mesh particularly congested, and perhaps could be used in future as part of the routing algorithm (probably not until v3) for avoiding busier congested areas if there are multiple routes to the destination?

SubnetConflict, 2024-04-15:
> This is a security problem that will help melicious actors, exposing
> firmware version makes it very easy to target specific problems.

That is a very valid point, it's a shame malicious actors have to ruin it
for the rest of us. But thank you for clarification.