meshtastic/firmware #3623: [Bug]: T-Deck Delay on typing

KyleMaas, 2024-04-15:
### Category

Other

### Hardware

T-Deck

### Firmware Version

2.3.3

### Description

When typing out a new message on the T-Deck keyboard, there is a substantial delay between when each key is pressed and when it registers and shows up on screen.  If you start typing another key before the first one shows up on screen, the intervening key presses are ignored.  This results in needing to type each key individually at a very slow pace if you don't want letters dropped.

thebentern, 2024-04-15:
This is a hardware limitation

closed 2024-04-15 by thebentern

KyleMaas, 2024-04-15:
@thebentern Could you explain a little bit as to why this is a hardware limitation?  CPU-bound?  Screen refresh taking so long that keyboard scanning can't happen?

KyleMaas, 2024-05-10:
So then is this considered a hardware issue because the delay is actually coming from the keyboard microcontroller being slow to process keypresses?

KyleMaas, 2024-11-15:
It's fixed in the new device-ui stuff - keyboard's immediately responsive.  So I think when that hits the mainline this should be fixed.