meshtastic/firmware #3624: [Bug]: If a message fails to send while you're typing a new message, the new message in-progress is cleared

KyleMaas, 2024-04-15:
### Category

Other

### Hardware

T-Deck

### Firmware Version

2.3.3

### Description

I believe this is most easily reproduced using the Long-Slow preset.

Type out a message, send it, and then start typing another message.  If the first message fails to send (like if you're out of range of any other nodes), the failed error message pops up on screen.  When that happens, it clears the message that was being typed so you have to type it again from the start.

KyleMaas, 2024-04-15:
I should note that I think Long-Slow makes this more frequent due to the amount of time it takes to send messages putting the error message squarely in the realm of where another message has already been started.

fifieldt, 2024-07-03:
@KyleMaas  Is this for Android, IoS, or Web app?

KyleMaas, 2024-07-04:
@fifieldt This is on the T-Deck itself.

HarukiToreda, 2024-08-28:
https://github.com/user-attachments/assets/e03695eb-02d4-4f2e-be1e-d42d84da7d1e

this is no longer an issue as seen this video, message received and it was not cleared

closed 2024-08-28 by thebentern