meshtastic/firmware #3628: [Board]: TWC_Mesh

SCWhite, 2024-04-15:
### SOC

NRF52

### Lora IC

SX1262

### Product Link

https://github.com/IISNRL/TWC_Mesh/tree/main/hardware

### Description

HI, This is a custom made version for local Meshtastic community.  
Our project goal is to provide the public with an open-source tool to face natural disasters or emergency situations.

This variant is based on Adafruit Feather nRF52840 Express + SX1262 Lora module + SSD1306 oled + ublox NEO-6M

[image]

[image]

Currently, the hardware and software has been tested with latest version (v2.3.4.ea61808)
shows compatible with latest function and didn't raise any critical error.

Please let me know if there is anything more I need to provide.

This is really a awesome project, thanks!

thebentern, 2024-04-19:
Hi! Feel free to create a pull request to add a new variant. Make sure to set the hw model to PRIVATE_HW. That board looks really neat and thanks for sharing!

closed 2024-04-19 by thebentern

SCWhite, 2024-04-22:
Thanks! will send a pull request ASAP
you can't imagine how excited I'm right now!