meshtastic/firmware #3635: [Feature Request]: Option to turn off LED inducators on ESP32 Heltec V3

punkpar, 2024-04-15:
### Platform

ESP32

### Description

Please can we have an option in the Android app to turn off the blinking indicator LEDS when connected to USB power. It lights up the window and room at night in which the Heltec V3 is in.

djoris, 2024-04-16:
in my opinion, that would be a great option.
Preferably the user could choose to have the LED always off, always blink, or blink when there's an unread message.
I'm not exactly sure how the different LED indicators work on different boards. As I see it now, the bright white LED on many Heltec boards just blinks every wake cycle or so, while other boards also have a power on LED, a GNSS LED...?

todd-herbert, 2024-04-16:
I suppose one function of the blink is that it indicates the crucial `OSThread` system is still running happily; not bogged down with one big blocking task. I'm not sure whether this was the original intention.

> It lights up the window and room at night in which the Heltec V3 is in.

I could certainly see the benefit in having a setting to disable the blink, for situations like this.

> [it could] blink when there's an unread message.

This could be another useful option, certainly on boards with only the one LED. There is already support for an alert LED, through the External Notification module. It would actually be fairly straight-forward to allow repurposing of the blink LED for alerts through this module. This isn't as user friendly as a simple toggle switch in settings would be, but it *would* avoid the hurdle of getting a new setting added, so it might make a good back-up option?

garthvh, 2024-04-16:
Problem is that we generally only control one or two of the up to 4 led's.

todd-herbert, 2024-04-16:
Going to have to implement at an even lower hardware level:
[image]

Sniffleupagus, 2024-04-16:
I implemented this last week as a user configuration option, and stalled on "now how do I check this in since it depends on a change to protobuf, and that seems kind of major..."  I was imagining "dark hours" to not blink during certain (or all. or none if you're into that sort of thing) hours, instead of just a. boolean, but so far I only implemented the boolean.

Is the hurdle of adding a config option a big one? And how would one go about such a thing?

todd-herbert, 2024-04-17:
> Is the hurdle of adding a config option a big one? And how would one go about such a thing?

From a technical standpoint:
* the protobufs are defined in https://github.com/meshtastic/protobufs/, that would be the place where a pull-request would need to be made.
* a good example commit showing a new setting added to the protobufs might be https://github.com/meshtastic/protobufs/commit/6ae27a2fbf288f7d984f048ee47c7ae71b24dd28
* what I have done previously for testing:
  * locally clone the git repos for firmware, protobufs and the python cli
  * make your changes in the protobufs repo
  * in both the firmware and python repo, change the protobufs submodule to point at your local protobufs repo
  * in both the firmware and python repo, use the scripts in the bin folder to rebuild their protobufs code. From memory the python script needed some tweaking (have been told that the python repo is a bit broken)
  * flash your device with the now updated protobufs code
  * use the python cli to test out changing your device's user config
* I assume that wider changes (apps, web-ui etc.) require manual intervention. See the matching Andoird App commit for the timezone protobuf change linked above: https://github.com/meshtastic/Meshtastic-Android/commit/4f29e7a3ffaef8aefaa7666ce4c8ded59764d6dd

From a decision making standpoint:
* my impression is that changes to user settings are more carefully considered, and that currently the general desire is to simplify user settings rather than add to them, though I do believe that new settings are still an option if the case is very strong.

AssetBurned, 2024-04-19:
Coming from yet another perspective… being able to switch off LEDs would also have a positive impact on battery life. Every little counts sometimes.

garthvh, 2024-04-20:
> Coming from yet another perspective… being able to switch off LEDs would also have a positive impact on battery life. Every little counts sometimes.

This is not going to have any appreciable battery life impact, and will not enable dark mode, all I see this doing is causing confusion

todd-herbert, 2024-04-20:
While the power savings would be negligible, it certainly sounds there's a demand for this as a customisation option.

ndoo, 2024-04-20:
Hi, I raised a draft PR to allow users to turn the LED off. https://github.com/meshtastic/firmware/pull/3674

Right now, the only option for those of us who have nodes we can see from our beds, or run a solar node and want the power saving, is to desolder/remove the status LED.

Also, for products like RAK4361 on RAK19007, the solar panel is connected to the battery charging IC, resulting in a 1Hz on-off cycle for the status LED to indicate charging, which does consume more power than the regular 1ms flash.

thebentern, 2024-04-20:
I don't see this as useful for power savings, but I do appreciate being able to turn off the Heltec white LED 😆

closed 2024-04-22 by thebentern

chichimus, 2024-07-31:
An app setting ti go dark would be great. In the meantime have a look at this thread. I updated my firmware and got the very bright flashing LED.  https://github.com/meshtastic/firmware/issues/4063

roboraptor, 2025-05-23:
Is it possible to just dim the LED? That would be ultrasuperawesome!
You can mount the LED inside the box, similar to how most of the fancy Heltec v3 setups have transparent buttons. However, those can get pretty bright at night—I'm not even talking about risking display burnout (angry). Relying on a timeout is just a workaround, not a real solution.

Dimming the blinking LED would help with:
- Avoiding being blinded by it
- Still being able to see it blink (which is important)
- Saving some energy

ndoo, 2025-05-23:
While it is possible to use Analog output to PWM the pin, not every pin supports it across all variants and MCUs.

roboraptor, 2025-05-23:
That is not the reason to not implement such a feature.

ndoo, 2025-05-23:
If you feel strongly about it, implement it and submit a PR, that's what I did. 

Or put a cash bounty on it.

brennanag, 2025-08-08:
Just a note to the devs, there is an option to turn off the blinking white led for my heltec v3 but it doesn't work. I had to use the CLI to do it. Others may find it confusing that the toggle on the app (iOS in my case) doesn't work. Cheers!

ndoo, 2025-08-08:
> Just a note to the devs, there is an option to turn off the blinking white led for my heltec v3 but it doesn't work. I had to use the CLI to do it. Others may find it confusing that the toggle on the app (iOS in my case) doesn't work. Cheers!

If you believe there is an issue with the iOS client, you can open an issue against https://github.com/meshtastic/Meshtastic-Apple

garthvh, 2025-08-08:
The heartbeat led switch works fine on iOS, you can't disable the battery led.

brennanag, 2025-08-08:
It actually doesn’t work for some. I am one of those. I’m glad it works for you.

garthvh, 2025-08-08:
> It actually doesn’t work for some. I am one of those. I’m glad it works for you.

Not really sure how this would be possible, the cli does the same thing.

brennanag, 2025-08-09:
I don’t understand it either. I was able to turn off the light using the cli but not on the app. Shrug.

ndoo, 2025-08-09:
> I don’t understand it either. I was able to turn off the light using the cli but not on the app. Shrug. 

Are you sure it isn't a bluetooth issue for you then?

dpuschek, 2025-08-25:
Hi there. I was actually able to disable the annoying white heartbeat LED on my Heltec Wireless Paper V1.1: 
Radio Configuration -> Device -> "Disable LED heartbeat - Controls the blinking LED on he device. For most devices this will control one of the up to 4 LEDs, the charger and GPS LEDs are not controllable"

 Now, however, the LED is completely off: Normally, it flashes brightly for 1 second when a new message arrives. Too bad. I like this indicator and was considering using a buzzer instead of the LED..  I like this indicator and was also considering using a buzzer instead of the LED at the PIN. Unfortunately, my Heltec Wireless Paper V1.1 doesn't have any other usable external pins. 
So, apparently, there's no "External Notification Module" in the firmware? The configuration menu under "Radio Configuration" is missing on this device! 

BTW: I used the webflasher and flashed Meshtastic Firmware 2.7.5.ddd1499 Alpha

dpuschek, 2025-08-26:
Thank you ndoo, I understand. So I have to do make a feature request to get the "External Notification Module" into firmware/variants/esp32s3/heltec_wireless_paper_v1 to get this notification feature?

dpuschek, 2025-08-26:
yes, it is actually missing. On my two other devices (rak4631 + heltec-wireless-tracker) I see a bell-icon labled "External Notification" under the "Module configuration" section.
I use the Android app, click the "three dots" on the top right corner and select "Radio configuration". Then I scroll down to the "Module configuration" section.

PR #3674 (2024-04-22): Add ability to turn off heartbeat LED blinking
Fixes #3635 and depends on [protobufs PR #485](https://github.com/meshtastic/protobufs/pull/485).

This only affects the GPIO-controllable LED and may be desirable behavior for Meshtastic users who sleep in the vicinity of their node and are annoyed by the blinking LED, especially for boards that indicate charging when USB power is connected, resulting in a 1 second LED illumination cycle.

Most boards however, have a non-GPIO controlled power LED (i.e. hardwired to VBUS/VCC/etc.) that can only be turned off by desoldering or otherwise removing th [... cut]
commit 250cf16b (2024-04-23): Add ability to turn off heartbeat LED blinking (#3674)

* Add ability to turn off status LED blinking

Fixes #3635 and depends on [protobufs PR #485](https://github.com/meshtastic/protobufs/pull/485)

* led_heartbeat_disabled

* trunk

---------