meshtastic/firmware #3641: [Bug]: NodeID Changed after Update

daniel-hofer, 2024-04-16:
### Category

BLE

### Hardware

T-Echo

### Firmware Version

2.3.5.2f9b68e

### Description

The NodeID changed after Update to latest alpha 2.3.5 on t-echo to !7c3b4eeb. Did nrf reset and resetted android app. Noticed, that my node was shown twice in the app and noticed, that the mac-accress nodeID changed. Noticed that the pairing started with 000_xxxx new node ID. 
Did a restore of the settings via app. Reinstalled android app and reflashed node.

GUVWAF, 2024-04-16:
So your initial flash was without full erase, right? 
Might it be that there was another node with this node number on your mesh? 
If you now reboot, I assume the node number stays the same. If you fully reset it again, will it have the original one or yet another new one?

daniel-hofer, 2024-04-16:
[image]
Node is showing zero as MAC-Address

GUVWAF, 2024-04-16:
Ah, so it's only the MAC address and thus BLE name, but not actually the node number that changed?

daniel-hofer, 2024-04-16:
Steps:
Save Node-Config via Android App Version 2.3.4
Full erase node with Meshtastic_nRF52_factory_erase_v2.uf2
Serial console and confirm factory reset with any key -> Formatting done -> node restart
Copy firmware-t-echo-2.3.5.2f9b68e.uf2 to node -> node restart
Delete local appdata on Android App.
Delete Bluetooth pairing
Search for node, node shows up with correct ble name Meshtastic_b561 after + Symbol.
Pairing with pin on display
Pairing done
Device Name Meshtastic b561 Region unset.
Device config -> factory reset from menu of android app -> reboot node
Now the Device Name changed to Meshtastic 0000 region unset.
Region set to EU_868 -> node reboot
Other Node hast now this node shown up as Meshtastic 0000 T_ECHO wit 00:00... MAC Address
 Node ID is !4162b561 in Settings, correct node.
Import config via Andorid app cfg file -> node reboot
After this, all went correct, but the shown mac-address in the nodelist from the webview of the other node still sowes zeroes for this node.
After delete the nodelist of the webview node, my device showed up with zeroes in the MAC Address section again.

daniel-hofer, 2024-04-16:
I made all steps again WITHOUT devicesettings/reset to factory settings from android app 2.3.4 and the mac address is shown correctrly again
[image]

GUVWAF, 2024-04-17:
Thanks for the elaborate explanation, I now understand what happened and it should not happen anymore with the above PR.

PR #3651 (2024-04-17): Fix #3641: Always set MAC when picking new NodeNum
Fixes #3641. Apparently the factory reset command first sets the long/short name based on the MAC address but it might be that `getMacAddr()` was never called. Since our NodeNum is only erased afterwards, it is not yet 0, so this part wouldn't be executed.
commit bc085ab8 (2024-04-17): Fix #3641: Always set MAC when picking new NodeNum (#3651)
closed 2024-04-17 by thebentern