meshtastic/firmware #3649: [Bug]: T-Echo Backlight not working on latest firmware

muchomustacho, 2024-04-17:
### Category

Hardware Compatibility

### Hardware

T-Echo

### Firmware Version

2.3.4.ea61808 Beta

### Description

Double clicking on the lower button doesn't turn on the backlight of the screen. This started since the screen sleeping feature was implemented. 

Putting the T-Echo to 2.3.2.63df972 Beta fixes the issue, any firmware newer that that will disable the backlight feature.

todd-herbert, 2024-04-17:
The standard behavior for Meshtastic devices is that double click sends an "adhoc ping" ("hello, I'm here!").
To separate the backlight from the adhoc ping feature, 4x click now toggle the backlight.

If you're interested, check out the discussion at https://github.com/meshtastic/firmware/issues/3533 and https://github.com/meshtastic/firmware/pull/3560.

muchomustacho, 2024-04-17:
Awesome, thank you for the quick reply!

closed 2024-04-17 by muchomustacho

GUVWAF, 2024-04-17:
Maybe we should adapt the page here: https://meshtastic.org/docs/hardware/devices/techo/buttons/
If I'm correct now also the capacitive touch button should be enabled with the "Wake screen on tap or motion" setting?

todd-herbert, 2024-04-17:
Good spotting!