meshtastic/firmware #3656: [Feature Request]: Meshtasticd Add Voltage measurement GPIO mapped to app device node voltage

ddave47, 2024-04-18:
### Platform

NRF52, ESP32

### Description

For portable or solar based mestasticd setups, it would be slick to have a voltage measurement pin that would be mapped to the node voltage field, in the app.  One could wire their battery to this voltage measurement pin so that battery voltage could be easily accessible. Battery voltage may be less than a PI VCC, like 4.2v if using a boost converter.  Or one could use a 12v SLA battery and the voltage field in the app would read 12v. This pin would be configurable in config.yaml.

garethhcoleman, 2024-04-21:
So most devices can probably only read up to 3ish V without using a potential divider - that is a pair of resistors. So you'd need to do some simple maths to work out how big each one needed to be depending if you wanted to read up to say 15V to cover a 12V lead acid battery. And this would need to be configurable by each user...

Do you think users who want this feature would be willing/able to wire in some resistors as well as the sense wire?

It's certainly do-able, but it's not super simple, especially as getting consensus to add more configuration options might be difficult.

ddave47, 2024-04-21:
I really like the idea of having this feature, but I don't think most user's would be willing to put in the effort to figure out the voltage divider circuit.  Possibly if there was an existing ADC chip that could be wired with minimal soldering connections, and the voltage was read automagically, it would be worth while.  May be hard to sell to dedicate resources at this time??

garethhcoleman, 2024-04-22:
I just realised that the current measurement devices INA260 and INA219 both also measure voltages and can cope with a 12V SLA battery no problem. So you might be able to get it working with something like this (Adafruit module)[https://www.adafruit.com/product/4226] for example.

closed 2024-04-28 by garthvh