meshtastic/firmware #3669: [Bug]: Unknown Timestamp on firmware > 2.2.23

thelamedia, 2024-04-19:
### Category

BLE

### Hardware

Rak4631, Heltec V3

### Firmware Version

2.3.2 and above

### Description

For devices that do not have RTC modules, the timestamp is no longer available beyond firmware 2.3.2.  When rolling back to 2.2.23, the timestamp returns.  Tested on Heltec v3 and Rak 4631 on multiple units.

garthvh, 2024-04-19:
A RTC by itself is not a source of time, what is your time source?

thebentern, 2024-04-19:
The only thing that could have possible affected this from the 2.2.24 release is the position precision feature's skipping of position broadcast for precision set to zero, in which case you should see this in the logs:
`Skipping location send because precision is set to 0!`

thelamedia, 2024-04-19:
My assumption is that the timestamp was previously being pulled from the Meshtastic app over BLE using the device GPS.  However after updating the firmware, none of the position settings transfer that data.

When looking at the position logs, Lat and Long are there, but timestamp shows as "unknown age"

thebentern, 2024-04-19:
Which app are you using, and can you grab some serial logs?

The web flasher now has a serial monitor if you need one https://flasher.meshtastic.org/

thelamedia, 2024-04-19:
> Which app are you using, and can you grab some serial logs?
> 
> The web flasher now has a serial monitor if you need one https://flasher.meshtastic.org/

I am using the iOS app.  I have pulled a few logs and don't see any position items listed, including the "skipping location send" mentioned above.  I have Channel 0 set to allow position and precise position is toggled.  Smart update is turned off and set to send every 5 minutes.  For the device GPS, settings are "not present" and "fixed position" is enabled.

thebentern, 2024-04-19:
> Here are some logs from a fresh reboot:

These logs didn't capture any position packets from the phone or the device. 

Since you're using GPS not present and fixed position, I'm assuming this is intended to be a well placed node in a static location, in which case the expectation is that the position remains "fixed" and the phone's position packets will only update the node's time. This is how it is _supposed to_ behave anyway.

thelamedia, 2024-04-19:
> Since you're using GPS not present and fixed position, I'm assuming this is intended to be a well placed node in a static location, in which case the expectation is that the position remains "fixed" [...]

Correct, but no time is updated in the current location, and even when moved I'm not seeing position or time updates. 

However in previous firmware it was working as intended.

PR #3689 (2024-04-21): Also refresh timestamp for "timeonly" fixed position nodes
Closes #3669
closed 2024-04-21 by thebentern