meshtastic/firmware #3675: [Bug]: Messages and ID not stored and lost in case of external power loss

Dorn8010, 2024-04-20:
### Category

Other

### Hardware

Heltec V3

### Firmware Version

2.3.4

### Description

Connected to a solar source, when the power is lost the device losses its last messages and sometimes its ID (long and short).

caveman99, 2024-04-20:
fixed in 2.3.6 and newer

closed 2024-04-20 by caveman99

Dorn8010, 2024-04-22:
Tired 2.3.6.7a3570a on Heltec V3.
Still loosing its short and long name and fixed pos.
Connected to a solar source with internal battery (2000mAh). 
After several restarts the name and pos are gone.

thebentern, 2024-04-22:
Sounds like you might have a hardware flash issue. I've seen this happen on NRF52 devices with dying flash.

Dorn8010, 2024-04-26:
Checked this issue on 2.3.6. with two other HeltecV3 boards. Same problem. But only after many restarts in a row. 
Losing ID (Long and Short) and the fixed position coordinates. The rest is still maintained e.g. Groups.