meshtastic/firmware #3687: [Bug]: adc voltage multiplier on rak11310 rak19007

quimnut, 2024-04-21:
### Category

Hardware Compatibility

### Hardware

Rak11310

### Firmware Version

2.3.6

### Description

The default adc multiplier on this platform is 3.1 which gives me readings over 7v for a lipo.

I have come pretty close, with 4.19v after the solar ends with it set to 1.85. 

I would like to report this and also see if anyone else can report experiences. thanks. 

### Relevant log output

```Shell
power:
adc_multiplier_override: 1.85
wait_bluetooth_secs: 1

Telemetry received:
Battery level: 99.00%
Voltage: 4.18 V
```

Donnie650, 2024-04-21:
I have 2 of these Rak 11310 units and both do same I have them  set to 1.84.

thebentern, 2024-04-21:
PRs welcome, if anybody wants to make the change.

PR #3698 (2024-04-22): adjust adc for rak11310 devices
Adjusts ADC on RAK11310 devices, fixes #3687 

The default of 3.1 reports over 7v.
closed 2024-04-22 by thebentern