meshtastic/firmware #3699: [Bug]: T-Echo Critical Fault #7 after changing Positions settings

muchomustacho, 2024-04-22:
### Category

Other

### Hardware

T-Echo

### Firmware Version

2.3.4.ea61808 Beta

### Description

I'm on the latest iOS app and the latest stable firmware (2.3.4.ea61808 Beta). When I import my channels and I go to change their position settings (make them Precise Location or change the Approximate Location) I get a **Critical Fault #7** error on the screen. 

I'm able to send and receive after I get this setting but when I restart the device, the error goes away but I'm unable to send any messages. They all time out. I can go and make a change on one of the channels and it gives the error again but then I'm able to send messages.

Only workaround I've come up for the error to go away and then keep the settings is set it up just the way I need it, then test I can send and receive and then do a reflash of the firmware, that seems to take care of it. 

[image]

caveman99, 2024-04-22:
"The channel tried to set a radio setting which is not supported by this chipset, radio comms settings are now undefined" - Please make sure the T-Echo actually fits the frequency range you wanna use and don't tinker with the radio settings too much. e.g. don't change frequency slot or offset if you don't know what you are doing.

Closing this, if problem persists even ond efault settings please provide debug serial logs and reopen.

closed 2024-04-22 by caveman99

muchomustacho, 2024-04-22:
The only change to the radio I made was the frequency slot so it matches the US; 20. I run my Long Fast channel as channel 1 instead of 0. Is that not supported by the T-Echo?

muchomustacho, 2024-04-22:
I've put it back to only Primary Channel and Frequency Slot to 0 and when I change the Position to Approximate Location 0.2 it gives the error again. 

How do I get you guys a log for that?