meshtastic/firmware #3700: [Feature Request]: Display INA3221 data on node info

geekfuns, 2024-04-22:
### Platform

ESP32

### Description

Hello, I have connected an INA3221 sensor to my ESP32 board and it is working well. Its measurement values can be seen in the debug information from the serial port and also the screen, but they cannot be seen in the app. The multi-channel measurement function of INA3221 is very useful for solar powered  stations, as it can read the voltage and current of the solar panel, as well as the working current of the system. May I ask if this feature can be added?
May the feature is already supported, can you tell me how to configure it?
Thanks a lot!

thebentern, 2024-04-22:
Which app? This the firmware repository. I can transfer this issue to the correct one

geekfuns, 2024-04-23:
> Which app? This the firmware repository. I can transfer this issue to the correct one

The mashtastic app.  Sorry, I posted this issue in the wrong place. Please help me move it over. Thank you

geekfuns, 2024-04-24:
I am using an Android application, and after replacing INA3221 with INA219, I can read the measurement values on the app.