meshtastic/firmware #3703: [Bug]: MQTT enabled with custom server/auth causes a reboot loop on T-deck

acidvegas, 2024-04-22:
### Category

WiFi

### Hardware

T-Deck

### Firmware Version

2.3.6.7

### Description

Enabling wifi and  mqtt with a custom mqtt server that uses auth and tls is causing the device to get stuck in a reboot loop.

oherrala, 2024-04-23:
@acidvegas Can you observe same things as documented in issue #3496 ?

acidvegas, 2024-04-26:
@oherrala not I cant observe anything. as soon as I set a custom mqtt server with tls and auth, it just loop reboots over and over

oherrala, 2024-04-26:
@acidvegas I don't know if there's documentation about debugging this stuff. I've been trying to figure these on my own. A quick guide:

Set `device.debug_log_enabled` to `true` (e.g. `mestastic --set device.debug_log_enabled true`) or from App.

After that you can open serial connection to the device via USB (e.g. `screen /dev/tty.usb<something>`) or try using the serial monitor on https://flasher.meshtastic.org/.

It should give you a log of what device is doing before it reboots.

acidvegas, 2024-05-05:
> @acidvegas I don't know if there's documentation about debugging this stuff. I've been trying to figure these on my own. A quick guide:
>
> Set `device.debug_log_enabled` to `true` (e.g. `mestastic [...]

The logs from the serial monitor stop at:
INFO  | 01:11:47 25 [WebServerThread] Attempting to connect directly to MQTT server mqtt.redacted.com, port: 1883, username: acidvegas, password: redacted

So I really can't debug past this point :(

acidvegas, 2024-05-05:
I should also mention the T3-S3's and Heltecs are having this same issue. So it's not even a T-deck related issue.

Is anyone using a custom MQTT server with TLS & auth successfully?

acidvegas, 2024-05-05:
I have gotten a connect to my MQTT server (finally, was an ACL issue in moquito), but TLS causes a reboot loop still

mkgin, 2024-08-30:
Same here... boot loop with TLS.  Then it started working. Unfortunately, I didn't have logcapture in minicom running when it started working at the time. Will see if I can reproduce and get logs.

setup:
- heltec paper v1.1
- firmware 2.4.2 beta
- TLS on port 8883 with let's encrypt certificates

mkgin, 2024-08-30:
> I should also mention the T3-S3's and Heltecs are having this same issue. So it's not even a T-deck related issue.
> 
> Is anyone using a custom MQTT server with TLS & auth successfully?
> 
> ALSO, you can fix your device by running `meshtastic --set mqtt.enabled false` RIGHT after it boots. This will stop the reboot loop so you don't have to reflash the device to make it stop.

Timing of this is challenging... I was unable to get the device into flashing mode anyway... I ended up blocking the device from the wifi network to get the loop to stop.

closed 2024-11-07 by github-actions[bot]

ufergus, 2025-01-18:
I am experiencing this condition on a t-beam supreme with 2.5.15 firmware.  Full erase and reconfiguration didn't resolve the issue.  Works fine with TLS disabled.  Once I enable TLS it will boot loop until I get a command timed right to turn it back off.  Haven't been able to get any further debug information from the device.  

Configuration was working for a while with an older 2.5.5 (i think) firmware but something triggered it to get stuck in the boot loop cycle.  SSL certificate, Traefik and Mosquitto containers were unchanged when it changed from functional to broken.  Still trying to track down what might have caused the change in behavior.