meshtastic/firmware #3716: [Bug]:

xxCLIMBINGMANxx, 2024-04-24:
### Category

Hardware Compatibility

### Hardware

Other

### Firmware Version

2.2.22

### Description

Hi there,

I am trying to create some mini meshtastic nodes with ESP32-S3 Mini and RF95 (aliexpress link: [ESP32 S3 Mini](https://www.aliexpress.com/item/1005006637349060.html?spm=a2g0o.detail.pcDetailTopMoreOtherSeller.10.55006f39qMk14K&gps-id=pcDetailTopMoreOtherSeller&scm=1007.40000.327270.0&scm_id=1007.40000.327270.0&scm-url=1007.40000.327270.0&pvid=f8433be8-32b3-4f8e-b85c-8606b20c0247&_t=gps-id:pcDetailTopMoreOtherSeller,scm-url:1007.40000.327270.0,pvid:f8433be8-32b3-4f8e-b85c-8606b20c0247,tpp_buckets:668%232846%238115%232000&pdp_npi=4%40dis%21CAD%218.59%218.59%21%21%2144.28%2144.28%21%402101e62517131503233007300ed67e%2112000037889966856%21rec%21CA%212015567574%21&utparam-url=scene%3ApcDetailTopMoreOtherSeller%7Cquery_from%3A)) but I can’t seem to get them to boot when I flash them with custom firmware. I keep getting an error on the serial port that says core dump flash is corrupted. Also the same firmware works on a different ESP32-S3 board and I have tested this board and it does indeed work. My platformio.ini file and the serial output is below. Any help would be appreciated

Thanks

```
Backtrace: 0x40377d76:0x3fceb180 0x403805fd:0x3fceb1a0 0x40387299:0x3fceb1c0 0x4206eab6:0x3fceb2f0 0x403776e5:0x3fceb320 0x403cd757:0x3fceb350 0x403cda16:0x3fceb380 0x403c9929:0x3fceb4b0 0x40045c01:0x3fceb570 |<-CORRUPTED

ELF file SHA256: da207282d3043e87

E (368) esp_core_dump_flash: Core dump flash config is corrupted! CRC=0x7bd5c66f instead of 0x0
Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x29 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40377984
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbe4
load:0x403cc700,len:0x2a38
```

```
[env:ESP32-S3-Nano]
board_level = extra
extends = esp32s3_base
board = esp32-s3-devkitc-1
board_flags = -DARDUINO_USB_CDC_ON_BOOT=1
build_flags = 
  ${esp32_base.build_flags} -D PRIVATE_HW -I variants/ESP32-S3-Nano
```

caveman99, 2024-04-25:
Sorry, we can't help with unknown board arrangements. Best bet for you would be to enable the stacktrace decoder in platformio and see where it crashes.

closed 2024-04-25 by caveman99

xxCLIMBINGMANxx, 2024-04-26:
I managed to figure out the issue on my own.
For future reference the issue was caused by platformio incorrectly assuming the esp32 chip has 8MB of flash when it only had 4MB so all i had to do was add the following lines to the platformio.ini file:
```
board_build.flash_size = 4MB
board_upload.flash_size = 4MB
```