meshtastic/firmware #3718: [Board]:

lordrak007, 2024-04-25:
### SOC

ESP32

### Lora IC

SX1276

### Product Link

https://resource.heltec.cn/download/Wireless_Stick_Lite

### Description

Hi, i have made varianf for older Heltec WSL. Now is officially supported only WSL v3.
I have tested WSL and works nicelly.

caveman99, 2024-04-25:
Thanks for your contribution. This device is deprecated already and it uses the older SX1276 chipset. I am hesitant to add the support to the official repository without someone willing to maintain it. Essentially all devices before Heltec V3 generation may still be in this repository but they're unsuppported. If a future firmware breaks them they will stay broken.

caveman99, 2024-04-25:
Sorry i closed this too fast. Giving other contributors the chance to voice their opinion on this.

lordrak007, 2024-04-28:
Hi i understand of your hesitate. It has standar ESP32 like several already supported devices, co why the are broken then older wireless stick and wireless stick duo stop working also. I understand that. I read many parts of mestashtics source code to try understand it and result is working WSL v2 (or 1).
Yes, i can build it myself every nebo update of meshtashtics but my goal is to be able update there devices by your web flasher and to be builded together with other variants.